CharlesGodwin/pymagnum #3: Problem with magtest

CharlesGodwin, 2019-11-18:
This is an issue reported on this [forum](http://midniteftp.com/forum/index.php?topic=2458.msg44179#msg44179) I have moved it here so all can view and or contribute.

`After a bit of messing about, I got it to spit out some info. But only my router.  My wires oriented like they are, i can read  magnum data with my scripts. I haven't flipped the wires to check with yours but, here is some of the data using your py program.

Options:cleanpackets=True, device='/dev/ttyUSB0', packets=50, timeout=0.001
Length: 5 UNKNOWN   =>FC000075BC
Length: 6 UNKNOWN   =>01013316221E
Length: 5 UNKNOWN   =>7301040002
Length: 4 UNKNOWN   =>5900FF00
Length: 6 UNKNOWN   =>00DA8A643C20
Length: 5 UNKNOWN   =>0000D2BE88
Length: 6 UNKNOWN   =>0014005A0078
Length: 3 UNKNOWN   =>0478A2
Length: 3 UNKNOWN   =>000001
Length: 6 UNKNOWN   =>FB000079BB01
Length: 6 UNKNOWN   =>013316221E73
Length: 5 UNKNOWN   =>0100000259
Length: 3 UNKNOWN   =>00FF00
Length:20 UNKNOWN   =>00DA8A643C200000D2BE880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:13 UNKNOWN   =>000001FB000079BB0100331622
Length: 5 UNKNOWN   =>1E73010000
Length: 4 UNKNOWN   =>025900FE
Length: 4 UNKNOWN   =>0000DA8A
Length: 6 UNKNOWN   =>643C200000D2
Length: 5 UNKNOWN   =>BE8800143C
Length: 6 UNKNOWN   =>3C00200000A4
Length: 4 UNKNOWN   =>000001FC
Length: 5 UNKNOWN   =>000079BB01
Length: 6 UNKNOWN   =>003316221E73
Packets:25 of 50 with 24 UNKNOWN, in 0.75 seconds
`

CharlesGodwin, 2019-11-18:
Try adding this option to the command line you used to run the test
`--timeout 0.005`
if that doesn't help increase it to 0.010
This option adds a delay to deciding when a packet completes. Your RS485 interface may be different from those used in testing. Please let me know what device you are using for RS485 and can I also have a list of your Magnum devices.

Watt-mike, 2019-11-18:
Might look at this, first. It took some messing about to get any data. It was so late last night when I was working on it.... anyway

 

Re: Reading Magnum Energy RS485 infromation

« Reply #3 on: Today at 03:01:42 AM »

QuoteModifyRemove

root@pi3bplus:/home/pi# pip3 install pymagnum-1.0.0.tar.gz
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./pymagnum-1.0.0.tar.gz
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (from pymagnum==1.0.0) (3.2.1)
Building wheels for collected packages: pymagnum
  Building wheel for pymagnum (setup.py) ... done
  Created wheel for pymagnum: filename=pymagnum-1.0.0-cp35-none-any.whl size=15505 sha256=ecad9ef85ff09efee9d41ff5216bf898b1ab5b567f5f5fe5d137cac2c3b2d2fe  Stored in directory: /root/.cache/pip/wheels/88/b3/66/732629d2cf6966399bfb55a5a1e52661ba15f5548cf51c44fb
Successfully built pymagnum
Installing collected packages: pymagnum
  Found existing installation: pymagnum 1.0.0
    Uninstalling pymagnum-1.0.0:
      Successfully uninstalled pymagnum-1.0.0
Successfully installed pymagnum-1.0.0
root@pi3bplus:/home/pi#

root@pi3bplus:/home/pi# python3 -m serial.tools.list_ports
bash: python3 -m serial.tools.list_ports: command not found

root@pi3bplus:/home/pi# python3 -m magnum.tools.magtest -d /dev/ttyUSB0
bash: python3 -m magnum.tools.magtest -d /dev/ttyUSB0: No such file or directory
root@pi3bplus:/home/pi#

If I mess about a bit, i can get data from the magnum setup but without knowing your code, I can't tell if it is reliable. 

Options:cleanpackets=True, device='/dev/ttyUSB0', packets=50, timeout=0.001
Length: 5 UNKNOWN   =>FC000075BC
Length: 6 UNKNOWN   =>01013316221E
Length: 5 UNKNOWN   =>7301040002
Length: 4 UNKNOWN   =>5900FF00
Length: 6 UNKNOWN   =>00DA8A643C20
Length: 5 UNKNOWN   =>0000D2BE88
Length: 6 UNKNOWN   =>0014005A0078
Length: 3 UNKNOWN   =>0478A2
Length: 3 UNKNOWN   =>000001
Length: 6 UNKNOWN   =>FB000079BB01
Length: 6 UNKNOWN   =>013316221E73
Length: 5 UNKNOWN   =>0100000259
Length: 3 UNKNOWN   =>00FF00
Length:20 UNKNOWN   =>00DA8A643C200000D2BE880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:13 UNKNOWN   =>000001FB000079BB0100331622
Length: 5 UNKNOWN   =>1E73010000
Length: 4 UNKNOWN   =>025900FE
Length: 4 UNKNOWN   =>0000DA8A
Length: 6 UNKNOWN   =>643C200000D2
Length: 5 UNKNOWN   =>BE8800143C
Length: 6 UNKNOWN   =>3C00200000A4
Length: 4 UNKNOWN   =>000001FC
Length: 5 UNKNOWN   =>000079BB01
Length: 6 UNKNOWN   =>003316221E73
Packets:25 of 50 with 24 UNKNOWN, in 0.75 seconds

And the error i got...

/usr/bin/python3: Error while finding module specification for 'magnum/tools/magtest.py' (AttributeError: module 'magnum/tools/magtest' has no attribute '__path__')

Thank you

CharlesGodwin, 2019-11-18:
I think your wires are right. If they are backwards you get a lot of FF bytes.

I have no idea why you're getting 
`(AttributeError: module 'magnum/tools/magtest' has no attribute 'path')`

I recommend one more attempt at a pip3 install
`sudo pip3 install --upgrade pymagnum`

if you are running as root do don't need the `sudo` part

But you do have the code running and using the right device. I'd like to hear about the timeout tests.

Watt-mike, 2019-11-18:
Charles

The code runs but it is not as you intended I'm sure. The error with the
/'s is because I have to point to the folders to find the program

Watt-mike, 2019-11-18:
The .tar.gz install was not complete, notice below. No answer to *which*

root@pi3bplus:/home/pi/pymagnum-1.0.0# which pymagnum
root@pi3bplus:/home/pi/pymagnum-1.0.0# which make
/usr/bin/make
root@pi3bplus:/home/pi/pymagnum-1.0.0#

CharlesGodwin, 2019-11-18:
I'm on new ground on this. I am not a pip expert and never tested the install from tar.gz. I've used the pypi.org instructions for posting a built system. I've relied on pip to publish with a simple `pip3 install pymagnum` and leaving it to find things. That seems to have been working. Try uninstall and a fresh install.

Of course you do have the program running so all this is optional.

CharlesGodwin, 2019-11-18:
I did a test of the tar.gz on a Pi that did not have pymagnum installed.
A log is attached

- at line 1, I execute a wget to retrieve the file
- at line 12, I execute the pip3 install of the retrieved file
- at line 22, I execute the module with the --help option

It seemed to work as expected
[pymagnum-install.txt](https://github.com/CharlesGodwin/pymagnum/files/3859209/pymagnum-install.txt)
.

Watt-mike, 2019-11-18:
Pip3 did work this morning, for whatever reason

CharlesGodwin, 2019-11-18:
That's good to hear. 
now to the real problem, reading the data correctly. ;)

Watt-mike, 2019-11-18:
I'm not sure what is going on. A different pi and a different location....

pi@watt:~ $ sudo pip3 install --upgrade pymagnum
-bash: sudo pip3 install --upgrade pymagnum: command not found
pi@watt:~ $ sudo apt install python3-pip
-bash: sudo apt install python3-pip: command not found
pi@watt:~ $

CharlesGodwin, 2019-11-18:
The bash message suggests that the command to bash is all one word. when I test bash with this it only echos the first word. your error reports the entire command line. Weird.
`pi@piserver:~ $ badcommand option1`
`-bash: badcommand: command not found`

Watt-mike, 2019-11-18:
pi@pi3bplus:~ $ python3 --version
Python 3.5.3
pi@pi3bplus:~ $ sudo pip3 install --upgrade pymagnum
-bash: sudo pip3 install --upgrade: command not found
pi@pi3bplus:~ $ sudo pip3 install pymagnum
-bash: sudo pip3 install pymagnum: command not found
pi@pi3bplus:~ $ sudo pip3 install 
-bash: sudo pip3 install : command not found
pi@pi3bplus:~ $ sudo pip3 
-bash: sudo pip3 : command not found
pi@pi3bplus:~ $ which pip3
/usr/local/bin/pip3
pi@pi3bplus:~ $ sudo /usr/local/bin/pip3 install --upgrade pymagnum
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pymagnum in /usr/local/lib/python3.5/dist-packages (1.0.0)
Requirement already satisfied, skipping upgrade: pyserial in /usr/lib/python3/dist-packages (from pymagnum) (3.2.1)
pi@pi3bplus:~ $

CharlesGodwin, 2019-11-18:
weird. What happens with `sudo which pip3`? Did you type the commands or cut and paste? I'm grasping at straws.

Watt-mike, 2019-11-18:
I updated the git site thing. Posted the commands and the reaults

Watt-mike, 2019-11-18:
Maybe I will get some time to make your program executable as stand alone

CharlesGodwin, 2019-11-18:
Does these work now?
`python3 -m serial.tools.list_ports`
and
`python3 -m magnum.tools.magtest --help`
or
`python3 -m magnum.tools.magtest -d /dev/ttyUSB0`

Watt-mike, 2019-11-18:
Yes. Only on the one pi i beat up

root@pi3bplus:/home/pi/pymagnum-1.0.0# python3 -m serial.tools.list_ports
/dev/ttyAMA0
/dev/ttyUSB0
2 ports found
root@pi3bplus:/home/pi/pymagnum-1.0.0# python3 -m magnum.tools.magtest
--helpusage: Magnum Test [-h] [-d DEVICE] [-n PACKETS] [-t TIMEOUT] [-nc]

Magnum RS485 Reader Test

optional arguments:
  -h, --help            show this help message and exit
  -d DEVICE, --device DEVICE
                        Serial device name (default: /dev/ttyUSB0)

seldom used arguments:
  -n PACKETS, --packets PACKETS
                        Number of packets to generate (default: 50)
  -t TIMEOUT, --timeout TIMEOUT
                        Timeout for serial read - float between 0 and 1
second
                        (default: 0.001)
  -nc, --nocleanup      Suppress clean up of unknown packets (default:
False)
Use `python -m serial.tools.list_ports` to identify devices.
root@pi3bplus:/home/pi/pymagnum-1.0.0# python3 -m magnum.tools.magtest -d
/dev/ttyUSB0
Options:cleanpackets=True, device='/dev/ttyUSB0', packets=50, timeout=0.001
Length: 4 UNKNOWN   =>200000A4
Length: 4 UNKNOWN   =>000001F9
Length: 6 UNKNOWN   =>00007BBE0100
Length: 5 UNKNOWN   =>3316201D73
Length: 6 UNKNOWN   =>010400025800
Length: 2 UNKNOWN   =>FF00
Length: 6 UNKNOWN   =>00DA8A643C20
Length: 5 UNKNOWN   =>0000D2BE88
Length: 6 UNKNOWN   =>001414870000
Length: 3 UNKNOWN   =>000011
Length: 2 RTR_91    =>9120
Length: 5 UNKNOWN   =>000001F900
Length: 9 UNKNOWN   =>007BBE01013316201D
Length: 9 UNKNOWN   =>73010400025800FF00
Length:20 UNKNOWN   =>00DA8A643C200000D2BE88001400000000000000
Length:17 UNKNOWN   =>000001F900007BBE01013316201D730104
Length:11 UNKNOWN   =>00025800FF0000DA8A643C
Length:15 UNKNOWN   =>200000D2BE8800140C190000B50080
Length: 7 UNKNOWN   =>815913B8FFFA12
Length: 5 UNKNOWN   =>271707FF3E
Length: 9 UNKNOWN   =>FFFF03F50A01000001
Length: 8 UNKNOWN   =>F900007BBE010033
Length:14 UNKNOWN   =>16201D73010400025800FF0000DA
Length:18 UNKNOWN   =>8A643C200000D2BE8800140C1914007300A0
Length:16 UNKNOWN   =>000001F900007CBD01003316201D7301
Packets:25 of 50 with 24 UNKNOWN, in 0.95 seconds
root@pi3bplus:/home/pi/pymagnum-1.0.0#

Watt-mike, 2019-11-18:
Also, with the delay

root@pi3bplus:/home/pi/pymagnum-1.0.0# python3 -m magnum.tools.magtest
--timeout .005 -d /dev/ttyUSB0
Options:cleanpackets=True, device='/dev/ttyUSB0', packets=50, timeout=0.005
Length:21 INVERTER  =>000001F900007ABE01013316201D73010400025800
Length:21 REMOTE_A3 =>0000DA8A643C200000D2BE880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>000001F900007ABE01003316201D73010400025800
Length:21 REMOTE_A4 =>0000DA8A643C200000D2BE8800143C3C00200000A4
Length:21 INVERTER  =>000001F900007ABE01003316201D73010400025800
Length:21 REMOTE_11 =>0000DA8A643C200000D2BE88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>000001F9000079BD01013316201D73010400025800
Length:21 INVERTER  =>0000DA8A643C200000D2BE88001400000000000000
Length:21 INVERTER  =>000001F9000079BD01013316201D73010400025800
Length:21 REMOTE_80 =>0000DA8A643C200000D2BE8800140C1B0000B50080
Length:18 BMK_81    =>815913B7FFFA12271707FF3EFFFF03F50A01
Length:21 INVERTER  =>000001F9000079BD01003316201D73010400025800
Length:21 REMOTE_A0 =>0000DA8A643C200000D2BE8800140C1B14007300A0
Length:21 INVERTER  =>000001F9000079BD01003316201D73010400025800
Length:21 REMOTE_A1 =>0000DA8A643C200000D2BE880014600090787878A1
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>000001F9000079BD01013316201D73010400025800
Length:21 REMOTE_A2 =>0000DA8A643C200000D2BE880014005A00780478A2
Length:21 INVERTER  =>000001F9000079BD01013316201D73010400025800
Length:21 REMOTE_A3 =>0000DA8A643C200000D2BE880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>000001F9000079BD01003316201D73010400025800
Length:21 REMOTE_A4 =>0000DA8A643C200000D2BE8800143C3C00200000A4
Length:21 INVERTER  =>000001F9000079BD01003316201D73010400025800
Length:21 REMOTE_11 =>0000DA8A643C200000D2BE88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>000001F900007CBE01013316201D73010400025800
Length:21 INVERTER  =>0000DA8A643C200000D2BE88001400000000000000
Length:21 INVERTER  =>000001F900007CBE01013316201D73010400025800
Length:21 REMOTE_80 =>0000DA8A643C200000D2BE8800140C1B0000B50080
Length:18 BMK_81    =>815913B8FFFA12271707FF3EFFFF03F50A01
Length:21 INVERTER  =>000001F900007CBE01003316201D73010400025800
Length:21 REMOTE_A0 =>0000DA8A643C200000D2BE8800140C1B14007300A0
Length:21 INVERTER  =>000001F900007CBE01003316201D73010400025800
Length:21 REMOTE_A1 =>0000DA8A643C200000D2BE880014600090787878A1
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>000001F900007CBE01013316201D73010400025800
Length:21 REMOTE_A2 =>0000DA8A643C200000D2BE880014005A00780478A2
Length:21 INVERTER  =>000001F900007CBE01013316201D73010400025800
Length:21 REMOTE_A3 =>0000DA8A643C200000D2BE880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>000001F900007CBE01003316201D73010400025800
Length:21 REMOTE_A4 =>0000DA8A643C200000D2BE8800143C3C00200000A4
Length:21 INVERTER  =>000001F900007CBE01003316201D73010400025800
Length:21 REMOTE_11 =>0000DA8A643C200000D2BE88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>000001FA00007CBE01013316201D73010400025800
Length:21 INVERTER  =>0000DA8A643C200000D2BE88001400000000000000
Packets:50 in 2.66 seconds
root@pi3bplus:/home/pi/pymagnum-1.0.0#

Watt-mike, 2019-11-18:
All 4 inverters are connected but only 2 are active, at the moment. I also
have a couple of device off, like the ags-n and an additional display.
There is s battery monitor active and a web module. Some other stuff but
can't remember

CharlesGodwin, 2019-11-18:
Try this one now for a JSON dump of your devices.
`python3 -m magnum.tools.testdump --timeout 0.005`

Your battery monitor should have shown up. Obviously the ags won't show up unless it is plugged in.

There may be a problem with multiple inverters. Another user reported a problem of not seeing what he expected but then had to shut down for the winter and I've no updates. If that is the case with you, please open a separate issue for the multiple inverters as it is a separate and highly specialized problem.

Thanks for your feedback.

Watt-mike, 2019-11-18:
It already sees both inverters that are connected and the bmk

Watt-mike, 2019-11-18:
root@pi3bplus:/home/pi/pymagnum-1.0.0# python3 -m magnum.tools.testdump
--timeout 0.005
/usr/bin/python3: No module named magnum.tools.testdump
root@pi3bplus:/home/pi/pymagnum-1.0.0#

boB-K7IQ, 2019-11-18:
You all have the Magnum document for their communications, right ?  Not sure if this is newer or not...

I know that I can insert a link but evidently cannot attach a PDF ?   Just in case, here is a copy/paste.

Last Revised: 18th Dec 2007
7-May-03 Update 10-Oct-03
Update 23-May-05 add 0% charge rate and 250 VAC Dropout for ac disconnect for EMS support
Update 28 March 2006 for new fault codes and new VAC dropout values
Update Oct 06 for sending VACin and VAC output and new fault codes:
RELAY_FAULT
CHARGER_FAULT
Update Nov 06 for new battery sizes and new TCO FAULT
Update Jan 23-07 added EOF (0xFF) value and third party notes.
Update Apr-23-07 added (Ox91) CB3 open fault
Update Aug-13-07 corrected VAC dropout values
Updated Jan-08-08 Added Custom battery charge info that RC send to INV
Owned by Magnum Energy, Permission is required to use this document and this protocol. Copyright 2003-2007 Magnum Energy Inc, WA
Software definition of comm system:
This is a definition of the "TCS" (Temp Comm Sys) for the "Charger"
part of the Magnum Energy Inverter/Charger/Remote pair.
TCS-C is the "Charger/Inverter" end.
TCS-R is the "Remote" end.
OVERVIEW:
This implementation works as:
"Charger/Inverter" is the MASTER of communications, and will
transmit its data once every 100ms. During this time
TCS-C will:
1. Enable the RS-485 transmit driver
2. (internally) disable the Receiver (RE)
3. Reset the RECEIVE DATA POINTER
4. Reset the TRANSMIT DATA POINTER, and transmit
16 bytes of data.
At the "end" of the transmission:
1. Disable the RS-485 transmit driver
2. (internally) ENABLE the Receiver (RE)
On reception of bytes (from the Remote):
1. Store data sequentially
2. advance the RECEIVE DATA POINTER
3. after the 16th byte,
(internally) disable the Receiver (RE).
(repeat)
"Remote" is the SLAVE and responds to messages by transmitting
its own message 10ms after the MASTER's message ends.
TCS-R operates in the following manner:
1. Receives bytes of data, storing to ram,
advancing the RAM_POINTER, and RESETTING
the Rx_CLOCK (2ms step) to ZERO.
When Rx_CLOCK reaches 5 (10ms) do:
1. (internally) disable Receiver (RE),
2. enable the RS-485 transmit driver
3. reset the TRANSMIT DATA POINTER
4. reset the RECEIVE DATA POINTER
5. trigger the TRANSMITTER
6. Every fourth send extends data transmission with AGS or BMK settings
When the transmitter is complete:
1. disable Transmitter
2. disable the RS-485 transmit driver
3. (internally) enable the Receiver (RE)
When AGS clock reaches 10ms, or BMK clock reaches 14ms:
1. "Remote" receives bytes of data, storing to ram
Note: AGS and BMK do not communicate on every cycle, only twice per second
(repeat)
Hardware definition of comm system:
This is a simple communications systems, where the physical
link is RS-485, which is defined as:
true 2 wire only
5V differential,
half-duplex,
multi-drop up to 32 connected transceivers.
RJ-11 wiring on inverter is (from top to bottom) :
[... 222 lines cut ...]
BMK Packet (data from the BMK)
Byte order Data name bits value assignments Value sent Comments
0 New Protocol header 0x8S S = String Type 8 means ME-BMK
1 State of Charge 8 0-100 - 0 - 100% or 1 count / 1% (255 = Think'n)
2,3 DC volts 16 0-64K 0-64K = 0 - 640.00V or 1 count/ 0.01V (real range = 0-70.00V)
4,5 DC amps 16 0-64K 0-64K = -3.2K to 3.2KAmps or 1 count/ 0.1A (real range +-1000A)
6,7 Min Volts 16 0-64K gVDC 0-64K = 0 - 640.00V or 1 count/ 0.01V (real range = 0-70.00V)
8,9 Max Volts 16 0-64K gVDC 0-64K = 0 - 640.00V or 1 count/ 0.01V (real range = 0-70.00V)
10,11 AmpH in/out 16 65536 0-64K = -32K to 32KAmpH or 1 count/ 1AmpH
14,15 AmpH Trip 16 0-64k = 0 to 6.4kAmpH or 1 count/ 0.1AmpH
12,13
Cumulative AmpH out
of Battery 16 0-64k = 0 to 6.4MAmpH or 1 count/ 100AmpH
16,17
Number of Discharge
Cycles 16 0-64k , 1 count / 1Discharge Cycle
18 Software Ver 8 0 to 256 Version
BMK TX every 0.5 seconds with a deadtime of 14msec after a ME-RC revision 2.0 or more
Notes for Third Party Developments
If your company is considering building a product to connect to the Magnum Network:
1 Please contact Engineering Department at Magnum for approval.
2 No power is available for third party devices on the network.
3 You should design your product so that it looks like a "wire" to the network. This is so that the timing of the data is not affected adversely and so that the devices on the
network will operate properly. Also, when your product is off or not operational, it will not affect existing communication on the network.
4 Magnum Energy Inc has rights to change/alter the protocol with or without notice to third parties

Watt-mike, 2019-11-18:
>    5. Disable the RS-485 transmit driver
>    6. (internally) ENABLE the Receiver (RE)
>    7. Store data sequentially
>    8. advance the RECEIVE DATA POINTER
>    9. after the 16th byte,
>    10. Receives bytes of data, storing to ram,
>    11. (internally) disable Receiver (RE),
>    12. enable the RS-485 transmit driver
>    13. reset the TRANSMIT DATA POINTER
>    14. reset the RECEIVE DATA POINTER
>    15. trigger the TRANSMITTER
>    16. Every fourth send extends data transmission with AGS or BMK
>    17. disable Transmitter
>    18. disable the RS-485 transmit driver
>    19. (internally) enable the Receiver (RE)
>    20. "Remote" receives bytes of data, storing to ram

Bob, midnite solar Bob?

CharlesGodwin, 2019-11-18:
I have a 2013 version of the doc, as a PDF. Thanks

CharlesGodwin, 2019-11-18:
@Watt-mike it should be
 python3 -m magnum.tools.**magdump**

sorry

Watt-mike, 2019-11-18:
I only have the 2007 as Bob has. Send me the 2013 pdf. Would you

Watt-mike, 2019-11-18:
No problem, I like Bob, very smart man

CharlesGodwin, 2019-11-18:
I can't send the 2013 version as I received it as a confidential document from a third party but I do have a 2012 version I have attached
[Magnum Networking Communications Protocol (12-16-10).pdf](https://github.com/CharlesGodwin/pymagnum/files/3860384/Magnum.Networking.Communications.Protocol.12-16-10.pdf)

Watt-mike, 2019-11-18:
I should have looked for the module. I only noticed 4 but I just been flat
out and can't really look into your code to see what's going on

root@pi3bplus:/home/pi/pymagnum-1.0.0# python3 -m magnum.tools.magdump
--timeout 0.005
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/pymagnum-1.0.0/magnum/tools/magdump.py", line 154, in
<module>
    magnumReader = magnum.magnum.Magnum(device=args.device,
packets=args.packets,
AttributeError: module 'magnum' has no attribute 'magnum'
root@pi3bplus:/home/pi/pymagnum-1.0.0#

CharlesGodwin, 2019-11-18:
try running as regular user and not root. i'm just guessing

Watt-mike, 2019-11-18:
Ok, well that stinks. I'll move on.thanks

Watt-mike, 2019-11-18:
I'll try, but the files have root permissions

Watt-mike, 2019-11-18:
Nah, same as.

Thanks for the information.

If I come across anything, I'll let you know.

Kinda hard to work with the devices with outdated facts.

Best regards
Mike

CharlesGodwin, 2019-11-18:
I don't know what to say about file execution problems. You seem to be having bad luck. what version of Raspian are you using? `cat /etc/os-release`

Concerning the manuals, the only change in the 2013 version is info on the PT-100 device. I don't think their documentation or product line changes a lot from year to year.

Watt-mike, 2019-11-18:
I have a pt-100, as I said.

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"

CharlesGodwin, 2019-11-18:
rerun magtest with `-n 100`. The PT-100 is shy about sending messages so capturing more packets at once can pick up intermittent packets. I'm surprised the BMK isn't showing up as a packet. .or do the same with magdump

That a good Raspian version, nothing strange there. I only recently changed up to the newer version. I'm still bewildered by the failures.

Watt-mike, 2019-11-18:
You may have missed where I mentioned the bmk *is* showing up.

Watt-mike, 2019-11-18:
How about I get another pi and install the latest Raspbian flavor on it.
Having more than one python 3 flavor on it gets a bit overwhelming for me
to keep track of. I'll do some messing about and get back to you.

Thanks

CharlesGodwin, 2019-11-18:
Ok.

Some background info

I have documentation on the pt-100 but no hands on access. What little info I have suggests it doesn't send a lot of packets. That's why I suggested you increase your packets size.

I'm not sure what is the story with multiple inverters. The doc suggests that a consolidated packet of data is sent but I have no way of knowing

I need someone to give me a full scale, maxed out system to test with ;)
Again, if it comes up, please open separate issues for each of these.

Thanks for your input. I learn more from problems than easy installs.

Charles Godwin

Watt-mike, 2019-11-18:
Charles, I have no idea how to create tickets or even what github is. I
have just been replying via email

CharlesGodwin, 2019-11-18:
Ok. Sorry. I'll sort it in the morning, eastern time.

Charles Godwin

Watt-mike, 2019-11-18:
I have 2 magnum routers. I think the 2 routers need to be updated before I
can make use of the pt-100. I decided to put it away for a while and just
keep midnite classics. Once i ac couple with this Aurora GT inverter, I
will break the magnum pt-100 out. So for now, no pt-100 is connected.

I will power on the 4 inverters and see whst happens but, it appears the
two inverters connected now show as two separate inverters

Watt-mike, 2019-11-18:
Would it do you any good to have arc's or rc's connected as monitors

CharlesGodwin, 2019-11-18:
I don't understand. Sorry

Charles Godwin

Watt-mike, 2019-11-19:
> weird. What happens with `sudo which pip3`? Did you type the commands or cut and paste? I'm grasping at straws.

Same as above 
pi@pi3bplus:~ $ sudo which pip3
/usr/local/bin/pip3

Watt-mike, 2019-11-19:
Email....

CharlesGodwin, 2019-12-05:
This seems to be fixed. There is a pre-release version [here](https://pypi.org/project/pymagnum/1.1b1/).

I recommend you uninstall the previous version first.
`sudo pip3 uninstall pymagnum` or
`sudo pip uninstall pymagnum` depending on your system

Remember to install with a Pi use
`sudo pip3 install pymagnum==1.1b1`

closed 2019-12-05 by CharlesGodwin