CharlesGodwin/pymagnum #4: Mis-identified packets.

CharlesGodwin, 2019-11-19:
In the attached file taken from a run of magtest program, there are three records identified as INVERTER. Lines 10, 30 and 50. But they are not consistent with the definition of an inverter record, they appear to be REMOTE records. 

If anyone has a chance, please run off more records to help with my testing. run the command

`python3 -m magnum.tools.magtest -n 100` followed by other options.

This will product a test with 00 packets. Add a file with the displayed packets to this issue.

Thanks

[misidentifiedpackets.txt](https://github.com/CharlesGodwin/pymagnum/files/3865205/misidentifiedpackets.txt)

CharlesGodwin, 2019-11-19:
A test release (1.1a3)  is available that fixes this problem. It is available for testing.

https://pypi.org/project/pymagnum/1.1a3/

Watt-mike, 2019-11-27:
Charles, after your update

pi@pi3bplus:~ $ python3 -m magnum.tools.magtest --timeout .005
Options:cleanpackets=True, device='/dev/ttyUSB0', packets=50, timeout=0.005
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025900
Length:21 REMOTE_A4 =>0000DA8A643C200000D2BE8800143C3C00200000A4
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025900
Length:21 REMOTE_11 =>0000DA8A643C200000D2BE88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025900
Length:21 INVERTER  =>0000DA8A643C200000D2BE88001400000000000000
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025900
Length:21 REMOTE_80 =>0000DA8A643C200000D2BE880014173A0000B50080
Length:18 BMK_81    =>815F13F2FFFA12271707FFAAFFFF03F70A01
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025900
Length:21 REMOTE_A0 =>0000DA8A643C200000D2BE880014173A14007300A0
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025900
Length:21 REMOTE_A1 =>0000DA8A643C200000D2BE880014600090787878A1
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025900
Length:21 REMOTE_A2 =>0000DA8A643C200000D2BE880014005A00780478A2
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025900
Length:21 REMOTE_A3 =>0000DA8A643C200000D2BE880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025900
Length:21 REMOTE_A4 =>0000DA8A643C200000D2BE8800143C3C00200000A4
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025900
Length:21 REMOTE_11 =>0000DA8A643C200000D2BE88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000077BC010033151E1C73010700025900
Length:21 INVERTER  =>0000DA8A643C200000D2BE88001400000000000000
Length:21 INVERTER  =>00000200000077BC010033151E1C73010700025900
Length:21 REMOTE_80 =>0000DA8A643C200000D2BE880014173A0000B50080
Length:18 BMK_81    =>815F13F2FFFA12271707FFAAFFFF03F70A01
Length:21 INVERTER  =>00000200000077BC010133151E1C73010700025900
Length:21 REMOTE_A0 =>0000DA8A643C200000D2BE880014173A14007300A0
Length:21 INVERTER  =>00000200000077BC010133151E1C73010700025900
Length:21 REMOTE_A1 =>0000DA8A643C200000D2BE880014600090787878A1
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000077BC010033151E1C73010700025900
Length:21 REMOTE_A2 =>0000DA8A643C200000D2BE880014005A00780478A2
Length:21 INVERTER  =>00000200000077BC010033151E1C73010700025900
Length:21 REMOTE_A3 =>0000DA8A643C200000D2BE880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000077BC010133151E1C73010700025900
Length:21 REMOTE_A4 =>0000DA8A643C200000D2BE8800143C3C00200000A4
Length:21 INVERTER  =>00000200000077BC010133151E1C73010700025900
Length:21 REMOTE_11 =>0000DA8A643C200000D2BE88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000077BC010033151E1C73010700025900
Length:21 INVERTER  =>0000DA8A643C200000D2BE88001400000000000000
Length:21 INVERTER  =>00000200000079BC010033151E1C73010700025900
Length:21 REMOTE_80 =>0000DA8A643C200000D2BE880014173A0000B50080
Length:18 BMK_81    =>815F13F2FFFA12271707FFAAFFFF03F70A01
Packets:50 in 2.68 seconds
pi@pi3bplus:~ $

Then with 100 count

pi@pi3bplus:~ $ python3 -m magnum.tools.magtest --timeout .005 -n 100
Options:cleanpackets=True, device='/dev/ttyUSB0', packets=100, timeout=0.005
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025900
Length:21 REMOTE_11 =>0000DA8A643C200000D2BE88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025900
Length:21 INVERTER  =>0000DA8A643C200000D2BE88001400000000000000
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025900
Length:21 REMOTE_80 =>0000DA8A643C200000D2BE88001400000000B50080
Length:18 BMK_81    =>815F13F2FFFA12271707FFAAFFFF03F70A01
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025900
[... 68 lines cut ...]
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025800
Length:21 REMOTE_A4 =>0000DA8A643C200000D2BE8800143C3C00200000A4
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025800
Length:21 REMOTE_11 =>0000DA8A643C200000D2BE88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025800
Length:21 INVERTER  =>0000DA8A643C200000D2BE88001400000000000000
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025800
Length:21 REMOTE_80 =>0000DA8A643C200000D2BE88001400000000B50080
Length:18 BMK_81    =>815F13F3FFFA12271707FFAAFFFF03F70A01
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025800
Length:21 REMOTE_A0 =>0000DA8A643C200000D2BE880014000014007300A0
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025800
Length:21 REMOTE_A1 =>0000DA8A643C200000D2BE880014600090787878A1
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025800
Length:21 REMOTE_A2 =>0000DA8A643C200000D2BE880014005A00780478A2
Length:21 INVERTER  =>00000200000078BC010033151E1C73010700025800
Length:21 REMOTE_A3 =>0000DA8A643C200000D2BE880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 INVERTER  =>00000200000078BC010133151E1C73010700025800
Length:21 REMOTE_A4 =>0000DA8A643C200000D2BE8800143C3C00200000A4
Packets:100 in 5.01 seconds
pi@pi3bplus:~ $

CharlesGodwin, 2019-11-27:
Thanks.
There are still some anomalies I'll fix before I release the next version.

Watt-mike, 2019-12-03:
Charles, installation just as shown.

pi@watt:~ $ sudo pip install pymagnum==1.1a3
Downloading/unpacking pymagnum==1.1a3
  Downloading pymagnum-1.1a3.tar.gz
  Running setup.py (path:/tmp/pip-build-5D9_4y/pymagnum/setup.py) egg_info for package pymagnum
    /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
      warnings.warn(msg)

    warning: no files found matching 'README.md'
    warning: no files found matching 'examples/maglogger.py'
Requirement already satisfied (use --upgrade to upgrade): pyserial in /usr/lib/python2.7/dist-packages (from pymagnum==1.1a3)
Installing collected packages: pymagnum
  Found existing installation: pymagnum 1.0.0
    Uninstalling pymagnum:
      Successfully uninstalled pymagnum
  Running setup.py install for pymagnum
    /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
      warnings.warn(msg)
      File "/usr/local/lib/python2.7/dist-packages/magnum/magnum.py", line 281
        raise unpack_error(msg) from e
                                   ^
    SyntaxError: invalid syntax

    warning: no files found matching 'README.md'
    warning: no files found matching 'examples/maglogger.py'
Successfully installed pymagnum
Cleaning up...

Watt-mike, 2019-12-04:
pi@pi3bplus:~ $ python3 -m magnum.tools.magtest --timeout .005 -n 100        Options:cleanpackets=True, device='/dev/ttyUSB0', packets=100, timeout=0.005
Length:21 INVERTER  =>400001EC000976B6010033152C1F73010003025800
Length:21 REMOTE_A0 =>0005DA8A643C200600D2FF880014011314007300A0
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC000976B6010033152C1F73010003025800
Length:21 REMOTE_A1 =>0005DA8A643C200600D2FF880014600090787878A1
Length: 6 AGS_A2    =>A20000000000
Length:21 REMOTE_00 =>400001EC000976B6010033152C1F73010003025800
Length:21 REMOTE_A2 =>0005DA8A643C200600D2FF880014005A00780478A2
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC000976B6010033152C1F73010003025800
Length:21 REMOTE_A3 =>0005DA8A643C200600D2FF880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 REMOTE_00 =>400001EC000976B6010033152C1F73010003025800
Length:21 REMOTE_A4 =>0005DA8A643C200600D2FF8800143C3C00200000A4
Length:21 REMOTE_00 =>400001EC000976B0010033152C1F73010003025800
Length:21 REMOTE_11 =>0005DA8A643C200600D2FF88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 REMOTE_00 =>400001EC000976B0010033152C1F73010003025800
Length:21 REMOTE_00 =>0005DA8A643C200600D2FF88001400000000000000
Length:21 REMOTE_00 =>400001EC000976B0010033152C1F73010003025800
Length:21 REMOTE_80 =>0005DA8A643C200600D2FF88001401130000B50080
Length:18 BMK_81    =>8155132AFF3812F615FBFEEE125200040A01
Length:21 REMOTE_00 =>400001EC000976B0010033152C1F73010003025800
Length:21 REMOTE_A0 =>0005DA8A643C200600D2FF880014011314007300A0
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC000976B0010033152C1F73010003025800
Length:21 REMOTE_A1 =>0005DA8A643C200600D2FF880014600090787878A1
Length: 6 AGS_A2    =>A20000000000
Length:21 REMOTE_00 =>400001EC000976B0010033152C1F73010003025800
Length:21 REMOTE_A2 =>0005DA8A643C200600D2FF880014005A00780478A2
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC000976B0010033152C1F73010003025800
Length:21 REMOTE_A3 =>0005DA8A643C200600D2FF880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 REMOTE_00 =>400001EC000976B0010033152C1F73010003025800
Length:21 REMOTE_A4 =>0005DA8A643C200600D2FF8800143C3C00200000A4
Length:21 REMOTE_00 =>400001EC000976B0010033152C1F73010003025800
Length:21 REMOTE_11 =>0005DA8A643C200600D2FF88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 REMOTE_00 =>400001EC000978AD010033152C1F73010003025800
Length:21 REMOTE_00 =>0005DA8A643C200600D2FF88001400000000000000
Length:21 REMOTE_00 =>400001EC000978AD010033152C1F73010003025800
Length:21 REMOTE_80 =>0005DA8A643C200600D2FF88001401130000B50080
Length:18 BMK_81    =>8155132BFF3612F615FBFEEE125200040A01
Length:21 REMOTE_00 =>400001EC000978AD010033152C1F73010003025800
Length:21 REMOTE_A0 =>0005DA8A643C200600D2FF880014011314007300A0
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC000978AD010033152C1F73010003025800
Length:21 REMOTE_A1 =>0005DA8A643C200600D2FF880014600090787878A1
Length: 6 AGS_A2    =>A20000000000
Length:21 REMOTE_00 =>400001EC000978AD010033152C1F73010003025800
Length:21 REMOTE_A2 =>0005DA8A643C200600D2FF880014005A00780478A2
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC000978AD010033152C1F73010003025800
Length:21 REMOTE_A3 =>0005DA8A643C200600D2FF880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 REMOTE_00 =>400001EC000978AD010033152C1F73010003025800
Length:21 REMOTE_A4 =>0005DA8A643C200600D2FF8800143C3C00200000A4
Length:21 REMOTE_00 =>400001EC000978AD010033152C1F73010003025800
Length:21 REMOTE_11 =>0005DA8A643C200600D2FF88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 REMOTE_00 =>400001EC00097AAE010033152C1F73010003025800
Length:21 REMOTE_00 =>0005DA8A643C200600D2FF88001400000000000000
Length:21 REMOTE_00 =>400001EC00097AAE010033152C1F73010003025800
Length:21 REMOTE_80 =>0005DA8A643C200600D2FF88001401130000B50080
Length:18 BMK_81    =>8155132AFF3812F615FBFEEE125200040A01
Length:21 REMOTE_00 =>400001EC00097AAE010033152C1F73010003025800
Length:21 REMOTE_A0 =>0005DA8A643C200600D2FF880014011314007300A0
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC00097AAE010033152C1F73010003025800
Length:27 UNKNOWN   =>0005DA8A643C200600D2FF880014600090787878A1A20000000000
Length:21 REMOTE_00 =>400001EC00097AAE010033152C1F73010003025800
Length:21 REMOTE_A2 =>0005DA8A643C200600D2FF880014005A00780478A2
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC00097AAE010033152C1F73010003025800
Length:21 REMOTE_A3 =>0005DA8A643C200600D2FF880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 REMOTE_00 =>400001EC00097AAE010033152C1F73010003025800
Length:21 REMOTE_A4 =>0005DA8A643C200600D2FF8800143C3C00200000A4
Length:21 REMOTE_00 =>400001EC00097AAE010033152C1F73010003025800
Length:21 REMOTE_11 =>0005DA8A643C200600D2FF88001414870000000011
Length: 2 RTR_91    =>9120
Length:21 REMOTE_00 =>400001EC00097AAE010033152C1F73010003025800
Length:21 REMOTE_00 =>0005DA8A643C200600D2FF88001400000000000000
Length:21 REMOTE_00 =>400001EC000978B3010033152C1F73010003025800
Length:21 REMOTE_80 =>0005DA8A643C200600D2FF88001401130000B50080
Length:18 BMK_81    =>8155132AFF3812F615FBFEEE125200040A01
Length:21 REMOTE_00 =>400001EC000978B3010033152C1F73010003025800
Length:21 REMOTE_A0 =>0005DA8A643C200600D2FF880014011314007300A0
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC000978B3010033152C1F73010003025800
Length:21 REMOTE_A1 =>0005DA8A643C200600D2FF880014600090787878A1
Length: 6 AGS_A2    =>A20000000000
Length:21 REMOTE_00 =>400001EC000978B3010033152C1F73010003025800
Length:21 REMOTE_A2 =>0005DA8A643C200600D2FF880014005A00780478A2
Length: 6 AGS_A1    =>A1013545007A
Length:21 REMOTE_00 =>400001EC000978B3010033152C1F73010003025800
Length:21 REMOTE_A3 =>0005DA8A643C200600D2FF880014602800200A00A3
Length: 2 RTR_91    =>9120
Length:21 REMOTE_00 =>400001EC000978B3010033152C1F73010003025800
Packets:100 of 100 with 1 UNKNOWN, in 4.79 seconds
pi@pi3bplus:~ $

Watt-mike, 2019-12-04:
The installed components are:
Router
Ags-n
Bmk
Magweb-wired
4 - MS4448PAE - inverters....
Parallel Stack Master configuration

CharlesGodwin, 2019-12-04:
I'm disappointed in the results. I've got work to do as now I've 'lost' inverter packets. all the REMOTE_00 packets that start with 4000 should be inverter packets.
I'll try to refresh today but can't commit

Thanks for your continuing support.

CharlesGodwin, 2019-12-05:
This seems to be fixed. There is a pre-release version [here](https://pypi.org/project/pymagnum/1.1b1/).

I recommend you uninstall the previous version first.
`sudo pip3 uninstall pymagnum` or
`sudo pip uninstall pymagnum` depending on your system

Remember to install with a Pi use
`sudo pip3 install pymagnum==1.1b1`

CharlesGodwin, 2019-12-06:
Release 1.1.0 is ready [here](https://pypi.org/project/pymagnum/).

If you have any version of this software installed I recommend you uninstall it using pip (or pip3) for Raspberry Pi users.
`sudo pip3 uninstall pymagnum -y`
`sudo pip3 install pymagnum`

closed 2019-12-06 by CharlesGodwin