CharlesGodwin/pymagnum #5: Multi-Inverters

CharlesGodwin, 2019-12-05:
There seems to be a problem with multi-inverters using the ME-ARTR. I recommended that the Pi connection use ports 5 or 6 of the router. I have had two users reports that only one inverter's data is showing up as a data packet. I have been reading the ARTR owners manual which I found [here](https://www.magnum-dimensions.com/me-artr-advanced-remote) and selecting it from the Downloads tab. Section 10 discusses setting up a remote ME-ARC remote and emphasizes that the remote can only display one inverter (referred to as a port). Which port you view is changeable, but only one at a time.

I'd like some help from anyone using this software. 

Could you extract packets, using magtest, to get inverter data packets. Then, using the ARTR panel select a different port for the ME-ARC, real of imaginary, and repeat the test. I suspect that you will get data from a different inverter. Repeat for each inverter in your system. Please let me now.

Now, what can be done about this? I suspect the solution is to use RS485/USB dongle for each inverter but I don't know enough yet to make a decision or any suggestions.

I encourage feedback. If you have data, please attach it a file to keep the messages simpler.

BTW I checked on the PT-100 situation and their design is different and I believe, since they are daisy chained, that the data is consolidated.

Watt-mike, 2019-12-06:
Charles, I do not have the advanced router. I have 2 of the older routers, an ME-ARC and an ME-RC. Sorry if I have confused you.

CharlesGodwin, 2019-12-06:
Not a problem. I'm just trying to gather information. Thanks for the update.

Watt-mike, 2019-12-06:
Charles, my cable is plugged into the master inverter through a splitter.

Watt-mike, 2019-12-06:
Does you program read and report all packets recognized or not of various
sizes?

CharlesGodwin, 2019-12-06:
My test program shows everything it reads on the RS485 interface. It does no filtering. As far as I know, no data is being missed. Unknown packets are marked as such. It does try to 'cleanup', or make sense of two consecutive unknown packets. That feature can be disabled if you are getting a lot of unknown packets by using the --nocleanup option in the command line. But no matter the settings all read bytes are displayed.

CharlesGodwin, 2019-12-06:
I've done more reading and the ME-ARTR and ME-RTR (older model), share a lot of features. I think my earlier discussion about settings/testing on the ARTR are valid for the RTR as well.

Another way to test is to connect the RS485 interface to the Network Port on the individual Inverters to see what you get. This may be harder as the ports on the inverters may be hidden by wires and covers.

Malkierian, 2020-03-28:
Hey, Charles, I have the ARTR and two inverters. I just barely got started with this, but wanted to chime in in the hopes of getting to understand it better (now that it's reading the network, though, I really have no idea what to do next as far as making that data readable).

Anyway, the ARC/RC can be set to SYSTEM as well as any individual port, and the SYSTEM setting is supposed to give an aggregate of the inverters' operation. So I ran magtest on SYSTEM, PORT1 and PORT2. I saw no changes at all in the place you were expecting (I'm guessing the other changes were packet types and information differences).
(log.txt) is without the ARC plugged in. (log2.txt) is with the ARC plugged in and monitoring PORT1, and (log3.txt) is with the ARC monitoring PORT2.
[log.txt](https://github.com/CharlesGodwin/pymagnum/files/4396498/log.txt)
[log2.txt](https://github.com/CharlesGodwin/pymagnum/files/4396502/log2.txt)
[log3.txt](https://github.com/CharlesGodwin/pymagnum/files/4396503/log3.txt)
I only have the ARC plugged in, so I have the RS485 adapter plugged directly into P5.

CharlesGodwin, 2020-03-28:
Thanks for your data. I've never seen anything like this. Is it possible the files were sorted, as packet types are usually interspersed rather than one type followed by another. 

That doesn't change the facts that I am seeing. Yes all inverter records appear to be the same. I have included a JSON dump of its contents as a file (I had to add .txt to get it uploaded)
[log.json.txt](https://github.com/CharlesGodwin/pymagnum/files/4397190/log.json.txt)

The concern I have isn't so much that the inverter data always seems to be the same but is it the aggregate data or just the 1st inverter? I suspect it's just the first inverter.  Run magdump, mentioned below, to get a better view of the data.

I wonder..... This may be difficult. Can you connect to the RS485 to the GREEN network port on each inverter and retest. It's voluntary and may be hard if your system is installed as they'd be hard to reach.

Concerning making the data readable. I recommend you extract the data as JSON. There is a program included, named magdump, that shows you how to do that.  There are also some programs in the examples folder on github. If you want to discuss this privately, my email is in the Feedback section of the documentation, or start a new issue on the subject.

Thanks for your help

Malkierian, 2020-03-28:
LOL, yes, I alphabetically sorted them to make comparing the packets easier, then saved them that way before uploading XD. If that's a problem I'll make sure not to do that in the future.

Here, log1 is the master inverter, log2 is the slave inverter. Also, these logs have all had 200 packets instead of 50.
[log1.txt](https://github.com/CharlesGodwin/pymagnum/files/4397473/log1.txt)
[log2.txt](https://github.com/CharlesGodwin/pymagnum/files/4397474/log2.txt)
And here's the dump, as you thought, it's only seeing the master the entire time, even when the remote is monitoring PORT2:
[dump.txt](https://github.com/CharlesGodwin/pymagnum/files/4397492/dump.txt)

CharlesGodwin, 2020-03-28:
The log1 and log2 files look like the wires are crossed. can you run again with a wire swap. Lots of FF in packets and unknown are usually a polarity problem.  Also where you plugged in, which P number on the router or if you plugged into the inverters directly.

Alas, the short coming of the dump program  is it only shows the last instance of each type of record. So if there were two packets from different inverters it wouldn't recognize the difference. This is the problem I'm trying to fix but need raw data to analyze. You are being very helpful.

Thank you, tank you, for your help.

Malkierian, 2020-03-28:
How odd, I didn't change the wiring between plugging it into the ARTR vs plugging it directly into the inverters. For the record, those two logs were directly into the inverters, and otherwise it's been in P5 the entire time. Interesting that changing between the inverters and the ARTR requires swapping the wires.

Anyway, here are the real logs. log1/dump1 for master, log2/dump2 for slave.
[dump1.txt](https://github.com/CharlesGodwin/pymagnum/files/4397583/dump1.txt)
[log1.txt](https://github.com/CharlesGodwin/pymagnum/files/4397584/log1.txt)
[dump2.txt](https://github.com/CharlesGodwin/pymagnum/files/4397585/dump2.txt)
[log2.txt](https://github.com/CharlesGodwin/pymagnum/files/4397586/log2.txt)
Looks like it can see the slave plugged into the slave, but you can't see both on an inverter's specific port.

Also, for the record, I figured out why it didn't seem to work on Windows. When you plug the RS485 cable into the ARTR, it takes a couple of minutes for the ARTR to actually recognize it's plugged in and start pumping data out to it, so I have to wait until I see the P5 LED light up before I can run pymagnum scripts.

CharlesGodwin, 2020-03-28:
@Malkierian I think Magnum used cross-over cables in some cases, so the pin out is different at each end of the cable. Why? I have no idea

Watt-mike, 2020-03-28:
Hello Charles,

I am still active with the Magnum stuff just not real active. I have been
spending a lot of time writing some code to help balance my 1800ah AGM
cells using MODBUS and the Midnite Classics along with some homebrew
hardware and some gear from Ross Wheeler.

If there is anything I can do to help provide some data, let me know.

Watt.

CharlesGodwin, 2020-03-29:
@Malkierian I made a long reply to you soon after you wrote including files dump1, log1, dump2 etc. But it has disappeared or I pushed the wrong button. Here goes again. I hope I remember all the important parts.

The data you sent was great. Thanks. It is obvious that each inverter is separate in these samples. I'm not sure if you sampled from the green port on the inverters or using a phone splitter at P1 and P2. If you didn't use P1 and P2 and you have a splitter, could you try. I ask as it's a lot easier for 
people to connect to the ARTR than to find the green port on the inverters. It would help my documentation.

- If this is true then users with 4 inverters would need to connect to a computer with the ability to handle 4 USB ports. A USB hub may work, it would depend on the hub and the computer. A Pi could do it but it would use up all USB connections. The Pi HAT solution can only handle 1 serial connection.

- The architecture of the pymagnum software will support one program running multiple instances provided each connects to its own serial port. The question is what to do about all the extra packets for the other peripherals as I assume they would be redundant on each inverter connection.

- What data is wanted? Trying to build a aggregate record in the published software is probably not a good idea as everyone would have their own view of "right". It's best left to the individual implementer.
- Would samples be worthwhile or ...

I learned something new about the ARTR not immediately connecting. Interesting.... I'll have to amend my documentation.

@Watt-mike Thanks for the offer. I'd appreciate anything you can send me. For each data sample tell me where you connected (inverter port on ARTR)? was it a split or just a connection? What other pieces are in the network (AGS, BMK, etc.)

I am now confident that the ability to read each inverter is possible. The question remains as to what is best AND simplest solution. I can think of several software solutions but each complicates the how to process. Perhaps just a supplemental document for multi-inverter implementers.

This is going to get interesting.

Watt-mike, 2020-03-29:
Charles, I have 4 - MS4448PAE inverters installed, if you remember. It may
be more practical for you to suggest the connections and the data you want
or need vs. what I can provide because I have no idea where you are going
with this.

Malkierian, 2020-03-29:
@CharlesGodwin Before I test that, I'm trying to determine if it's OK to hotswap the P1-P4 ports while the inverters are operating in stack. It's got a separate stack cable on there, so I assume the critical stuff is handled through there, but I can't be sure, and I don't want to damage anything. Do you know the answer to that?

CharlesGodwin, 2020-03-29:
@Watt-mike 
OK. I looked at the installation guide for RTR and it is very similar to the ARTR as far as cabling is concerned.
- Are you using both RTR with 2 inverters on each or just one RTR with 4 inverters?
- You said last year you used a splitter into the master inverter. Is the splitter plugged into the REMOTE (blue) port of the inverter? or?
- Have you tried plugging it in at the other end of the cable at P1 on the RTR? should be the same, polarity may be reversed. :(
- each test consists of running `magtest --packets 100 --device /dev/ttyUSB0` or modify to suit.
- If a test works for P1 with the splitter, can you repeat tests for P2, P3 and P4 and (P5 or P6). The last test is a reality check.
- If the P1 test doesn't work, can you test using the NETWORK (green) connector on each inverter.

Send me text files of the test outputs.

I'm asking a lot so if it's too much say so. Once I know what's possible and what Magnum is actually doing, I'll be able to help users like you and @Malkierian get data from all their inverters.

Thank you
Charles

Watt-mike, 2020-03-29:
I wouldn't swap them hot as the router assigns a master and cycles the
slaves as needed.

CharlesGodwin, 2020-03-29:
@Malkierian I defer to @Watt-mike  I have no experience with multiple inverters. that's why I' asking all these dumb questions.

Malkierian, 2020-03-29:
Well, that test will have to wait, then. I sent an email to Magnum with a hypothetical situation aimed at getting a response on that. I still can see it being a possibility, especially if I just unplug it and plug it into the splitter really quickly, but, if they give me an answer, they would know. If not, it'll have to wait until Monday when I can get another splitter and have them both plugged in at the same time after I power down the system.

Watt-mike, 2020-03-29:
Did I understand you correctly that you are swapping say p1 to p2?

Or, are you unplugging p1 adding a splitter and then plugging the splitter
back in to p1?

Malkierian, 2020-03-29:
Unplugging, adding splitter, and plugging it back into the newly split port.

Watt-mike, 2020-03-29:
Ahh, ok, I have definitely done that before. One time I did that it turned
the system off is all.

Malkierian, 2020-03-29:
So, then, the only risk is shutting the system down? And that's not guaranteed to happen?

Watt-mike, 2020-03-29:
Well, I'm not sure about that being the only risk. I have only had it shut
the system down once. The other times it was uneventful. I am not sure if
there are other problems possible, I am just explaining my experiences.

Malkierian, 2020-03-29:
Alright, I was able to get the splitters in without having to shut down, and I have more logs! Same association as before (1 = master, etc), and these are reading off of the remote cables from the inverters via splitters on P1 and P2:

[dump1.txt](https://github.com/CharlesGodwin/pymagnum/files/4398245/dump1.txt)
[log1.txt](https://github.com/CharlesGodwin/pymagnum/files/4398246/log1.txt)
[dump2.txt](https://github.com/CharlesGodwin/pymagnum/files/4398248/dump2.txt)
[log2.txt](https://github.com/CharlesGodwin/pymagnum/files/4398249/log2.txt)

Now, keep in mind, I had to use an old driver for one of my adapters (they're not the same kind) because it was an old, unsupported chip and was broken with the driver installed via Windows Update. I'm pretty sure that's what's causing the unknown packets on the slave logs (which was plugged into the faulty one), since I was able to switch the cables to my RS485s and get a clean packet read off of the slave through the good adapter. But I can access them both at the same time. I'm about to go through the process of creating a dual boot on my laptop, until I can get a Raspberry, so I can retain settings when I go on there to test this stuff out (because I think MX Linux was able to utilize the weird device just fine).

CharlesGodwin, 2020-03-29:
@Malkierian Thanks for all your effort. I really needed to know this stuff.

This looks like good news so far. 'We" can read network traffic of individual inverters on the P1-P4 range. That makes set up a lot easier for implementers as physically getting to the connectors on the inverters could be difficult.

Since you did your testing with nothing else connected, I still don't know if, or how, network traffic for components such as the Automatic Generator Start (AGS) and Battery Monitor (BMK) fit in. They may only be visible on P5 and P6. From your earlier tests it seems that P5/P6, also includes the inverter messages for the master inverter.

@Watt-mike may be able to shed light on this if he can complete some more tests. We shall see.

At this point I think I see a setup where the connections would be:
- P5 or P6, with or without a splitter as other wiring dictates, would collect all peripheral messages as well as Master inverter.
- P2 - P4, with a splitter or GREEN on the inverters, would pick up traffic from the slave inverters.

A data collection program would need to read data from 2 to 4 serial ports, one to match each described above. The data collected from P5/P6 would have all data for everything but slave. The data collected from the P2-P4 devices could have just the inverter data extracted and consolidated with the master data to be data point. P1 should be the same inverter data as P5/P6  - to be confirmed.

I think writing documentation to explain this and provide a code snippet will take longer than writing real code. But I think it's worth it.

I'll start writing something later today or Monday.

What's next for you. Ready to set up. If you want to discuss "what do I do with data", open a new issue or write me. My contact information is in the feedback section of he documentation https://pymagnum.readthedocs.io/en/latest/

Charles

Malkierian, 2020-03-29:
The one problem I see with that is the length of time it takes the ARTR to start sending network information over P5 without a splitter. I'm going to test splitting P6 with my remote to see if some different data is possible, and also see if I can test packet similarity for the inverter on both of them compared to P1. Also, I actually do have two BMKs, one plugged into each inverter. Any idea why they never showed up?

CharlesGodwin, 2020-03-29:
I read the installation guide for the ARTR and it shows the BMK plugged into P5/P6. It's figure 2.1. but the manual also says: (my emphasis)

> Info: A maximum of four inverters can be connected to a router to work in parallel with
each other. If the maximum number of inverters are connected, two open ports remain.
If a ME-ARC or ME-RC remote is connected to a router, it must be connected directly to
a remote communication port on the router. If there are still open communication ports
on the router, connect any accessories—i.e., an ME-AGS-N or ME-BMK—directly to the
router first. **Then, if all the router’s communication ports are being used and there are
additional accessories, they may be connected to the Network ports on the inverters**.

So the BMK plugged into GREEN ports on the inverters should work. I presume they're properly wired. Do they show up on the ARTR display panel?

I suspect the delay on the ARTR not responding to the RS485 reader is that the RS485 dongle never transmits, only receives.

CharlesGodwin, 2020-03-29:
How do the two BMK units work? Are the inverters connected to two battery banks, separate AC circuits? I just can't think of how two are used or needed. I learn every day.

Malkierian, 2020-03-29:
We figured that each inverter used the BMK locally, that's why we plugged them into the inverters directly the way they would have been for a single inverter by itself. I'll have to try plugging them into the ARTR instead, see what happens. Also, I think we have one BMK each on each layer of batteries (two shelves of them) so they're technically monitoring two banks separately. But I don't know if the ARTR can utilize more than one.

Watt-mike, 2020-03-29:
I'm surprised you don't have conflicts with having more than on BMK on the
system. I'm not sure how they could manage two separate banks.

Even the classics have trouble with more than one monitor, ie Whizbang Jr.

Malkierian, 2020-03-30:
@CharlesGodwin Is it possible to make the RS485s transmit signal, to try to emulate a remote or something and get it to communicate faster, or can they only receive?

CharlesGodwin, 2020-03-30:
Technically yes but not something I'm interested in doing. The risk of blowing out a system is too high

Charles Godwin

Malkierian, 2020-03-30:
That makes sense, but is unfortunate, since I've left a device plugged into P5 for up to an hour without getting any network output...

Also, for the record, the unknown packets were from a bad driver implementation. That device works perfectly on Linux. And, plugging an RS485 into the ARC channel does not work. My ARC registered a fault and required a restart.

Also, I'm not sure exactly what I have anymore, because I don't think I have a BMK, I only have the BTS that's plugged into each inverter. Trying to plug that into a network port, or P5, started the fault light flashing on the ARTR.

Watt-mike, 2020-03-30:
The BTS is most likely just a three wire ds18b20 sensor and the standard
seen it as a near short on the network bus. The BMK's are magnum network
and either wireless or cat5 network.

CharlesGodwin, 2020-03-30:
I think the BMK is a true Magnum energy device and uses the standard RJ11 (RS485) connection

CharlesGodwin, 2020-03-30:
Yes, plugging a BTS into the network could cause a fault. There is YELLOW socket on the inverters exclusively for the BTS. It is labelled Battery Temp Sensor. That's the only place to connect it.

Watt-mike, 2020-03-30:
You are right. My mistake.

I was thinking of the MagWeb. I also have a MagWeb that doesn't show up. I
suppose it strictly listens.

CharlesGodwin, 2020-03-30:
I haven't worked with one but I assume it only listens. Does cause P5 to light up right away or....

BTW, Magnum imply in their documentation that a Magweb can only report one inverter.

Watt-mike, 2020-03-30:
Charles, it reports more than one.

CharlesGodwin, 2020-03-30:
Fabulous. I thought from literature was one. In your case how does it distinguish them?

Watt-mike, 2020-03-30:
I will have to get back to you on that. I will see if it reports the router
status because it is plugged in to it. And, what status if it is only
plugged into the inverter

CharlesGodwin, 2020-03-30:
@Watt-mike 
Thanks this is all helping nail down a reliable way to report each / all inverters in a network using my software. The easier, the better.

The documentation lists these 5 stack mode values. I can infer the first three, but I have no idea what a Serial stack is. In you case with 4 inverters, do they all have different modes? or...?

<pre>
Standalone unit 	0x00  
Parallel stack - master	0x01  
Parallel stack - slave 	0x02  
Series stack - master 	0x04  
Series stack - slave    0x08
</pre>

Watt-mike, 2020-03-30:
Mine is in fact reporting 0x01

CharlesGodwin, 2020-03-30:
Interesting. For Amps values do you get a sum of all 4 inverters or just P1?

Watt-mike, 2020-03-30:
When I get home this evening, I will break down how that is reported.
Because I have a BMK, that current through the shut is mostly used

CharlesGodwin, 2020-03-30:
But the inverters do report Amps AC. Can that vary by device? If it’s all smoothed out I don’t need to worry about individual inverters.

I suffer from ignorance. The only multi-inverter experience I have is with my brother’s Outback units which work in sequence. First inverter gets loaded then the second one kicks in etc.

Watt-mike, 2020-03-30:
Calculated amps

Watt-mike, 2020-03-30:
Anything through the inverter is calculated amps

CharlesGodwin, 2020-03-30:
I assume calculated means total, all inverters, I might not have a multi inverters problem.  If P5 / P6 produce a calculated number, all this fussing is irrelevant. I hope

CharlesGodwin, 2020-03-30:
True but for Amps AC where else can you get it. Is it relevant?

Watt-mike, 2020-03-30:
Calculated even 1 inverter.  Only with bmk is the amperage accurate

Malkierian, 2020-03-30:
I will mention that, while it tries, it doesn't quite get entire even on the power generation distribution. I frequently have one inverter running 4-8 amps below the other.

Malkierian, 2020-03-30:
Also, the Magnum people got back to me on my question, and they say that there's no risk of damage or anything from even swapping the remote ports the inverters are plugged into, and it should update which one is actually the master and slave in the ports list.

ve2mxo, 2021-01-17:
Charles, my installation is out network with one ME-ARTR, 2 MS4448PAE (master/slave), one pt-100.
Using magdump, port 5 or 6 see only the master, i connect a second link to the remote port of the slave inverter : only one inverter is reported (the slave) and occasionnally the PT100 when i am lucky, but the PT100 is connected to the remote port the the master inverter. 
What is the best connection layout to get all information for all devices?

I also have stability issues with the magdump, ramdomly it does not work and go in error state, when it finally work (in a ten minutes loop) it fall in error again after some iterations.  How can i stabilize the magdump application (the same comments apply to magtest)?.

CharlesGodwin, 2021-01-17:
Multiple inverters are  a problem.  I have no hands on experience, but I think one RS485 device connected to port 5 or 6 to collect all data except the second inverter.  Then a second RS485 device for the second inverter only and ignore other data programmatically. You need to run separate instances of any program to read the second one. 

The PT-100 is not a talkative device. I think it only reports data when something changes and perhaps at midnight for daily resets. I wish I could be wiser.

I'll take a look at magdump to see if I can make it more robust. there are also example programs [here ](https://github.com/CharlesGodwin/pymagnum/tree/master/examples)that may be more useful. 
Can you give me any more information about the failure. Do you get messages or ....?

Watt-mike, 2021-01-18:
Charles, just an FYI, I have been working with the RV-C CANbus. I have been
using it to read and control inverters and such. It gets away from the
Display requirements, so far. I've only messed about with it connected to
the MSH3012M, battery monitor, N-genstart etc...

Malkierian, 2021-01-18:
> Charles, my installation is out network with one ME-ARTR, 2 MS4448PAE (master/slave), one pt-100.
> Using magdump, port 5 or 6 see only the master, i connect a second link to the remote port of the [...]

I had the same problem, only way I found out to do it was to have a separate serial device for each inverter tied into each data port through a splitter (so cable comes from inverter, into splitter, one split goes to RJ485, other goes into com port on ARTR). I then modified magdump to read from both devices at the same time in one cycle, so I didn't have to run separate instances for each inverter. There is no individual data coming out of p5 or p6, and having something spliced into the cable going to a second remote would sometimes cause the whole system to go haywire until unplugged.

CharlesGodwin, 2021-01-18:
thanks for the info/update. Where do you get the other devices' data. Such as ARTR, BMK, any others.

[5 more replies]

closed 2023-02-22 by CharlesGodwin