CharlesGodwin/pymagnum #8: Utilizing Pymagnum Data

Malkierian, 2020-03-29:
OK, now we get to the interesting part (I figured I'd do it on here so there's public access to the progress, in case someone else wants to do this as well). Here's the gist: I started into this to be able to have local, customized access to monitoring for the Magnums and my Classic 150s (especially the magnums, and without that ridiculously expensive MagWeb) . So eventually, this will need to be fed into a database that I can access easily from a webpage (I haven't even gotten started on the Classics yet simply because I can monitor them right now using the Classic Monitor app on Android, and I don't have any monitoring of the Magnums). That means there will be a server of some kind reading packets at every particular interval (not sure how big that interval will be yet) and writing an average of those packets received to the database. I'd also like to have it able to respond to internet protocol requests of some kind (TCP, HTTP, whatever) with the currently stored latest packet information for a live dashboard. It's quite possible the database might only have datapoints every 5, 15, or even 30 minutes, not quite sure (don't want to populate the database too quickly or densely).

Heck, even if, on your end, all that was made was something to broadcast that periodic data from all available devices on the given channel (JSON would be sufficient), or to give that information easily upon request (again, some web or socket protocol), and configure which RS485s to read from to do it at the same time, that would probably be sufficient.

CharlesGodwin, 2020-03-29:
Choices, so many choices. My choices were constrained by my lack of reliable and cheap internet service so I store all my data on my Pi attached to the Magnum System. It's a Pi 3 B+ with a 120GB SSD USB drive.

But consider the total setup and then you can decide what to do, and where. I can discuss the specifics in a subsequent reply.

I am using three software packages to do most of the heavy lifting. All running on the Pi. There is a little bit of setup and Grafana requires defining the display but no complex programs.

1. MQTT Server. Google MQTT for more information that will want to read. The short description is it allows me to publish (push) data to a MQTT broker (server) with a topic, AKA tag, attached. The program publishing doesn't care where it's going, it's just sent to the server, which will deal with it. More later. This is available as a Pi install or http://mqtt.org/ for other systems.

2. InfluxDB - This is a time series data store. It's not really a database in the traditional sense as you don't pre-define data fields etc. but it is very good at serving up timestamped clumps of data.
https://www.influxdata.com/ They sell commercial copies but also a community (free) version. Setup is simple.

3. Grafana This is a tool that implements a web browser interface  for displaying data in dashboards. The Design tool is the web and it's all point and click. I've attached a screen shot
!from July 5 /6 last year
[image]
. I keep my server in town over the winter.  https://grafana.com/

I am monitoring my Magnum network, a MorningStar MPPT charge controller via Modbus and temperature sensor attached to the pi.

These are monitored by three separate but similar programs. All code I write is in Python 3. They run a loop with sleep time of 60 seconds. I tried 30 seconds and I don't see any improvements in statistics and more that 60 seconds and I get impatient for a refresh. In each, the program gets the latest set of data points and publishes it to the MQTT broker and goes to sleep. In the github examples folder there is a program named [MQTTlogger.py](https://github.com/CharlesGodwin/pymagnum/tree/master/examples) that show an example of how this is done.

I have another program running, which is not included as an example, that is event driven. It connects to the MQTT broker and says "give me all messages for these topics (tags)". When a message arrives at this program it is dissected and posted to the InfluxDB data store.

That's it for programming. The next refresh of the Grafana web page and the latest data is presented. Grafana has an incredible number of mouse driven tools for zooming in, zooming out etc. 

All of these pieces can be distributed.  Obviously the Magnum monitoring program must be attached to the Magnum network but the MQTT broker can be at any reachable internet address, the same for the InfluxDB data and the Grafana server. I guess there could be multiples of all these but I've never considered it.

The simplicity of this is that not much programming effort has gone into digesting the data. The programs just get what's presents, post it to a database and then the graphic front end asked for what it wants.

I am not concerned with data storage. I just checked my server and the entire server with software is using less than 4GB and that includes over 4 months of data stored every 60 seconds. 

Now, you can do something different and more traditional. In the examples folder called magserver.py. It is a http REST implementation. Every time you send it an http request, it reads the current state of the Magnum network, and returns the data as a JSON string. You could write code to  call this every 60 seconds and do all your data logging etc. using traditional methods. There no right or wrong way to do it. 

I rote this code to only provide, as the base software, the core tool to extract the data. How and when it is extracted and exploited is up to the implementer. 

Now, the part about needing multiple RS485 dongles to deal with multiple inverters? I'm still mulling that one over. See https://github.com/CharlesGodwin/pymagnum/issues/5. I have several ideas. I'm trying to think of a simple but effective solution.

BTW The way the reader works is to sample the network traffic as quickly as possible, the Magnum data is like a fire hose, it just sprays data. My code grabs 50 packets (this is adjustable) and then processes it. In the case of duplicate packets the last one is used. The resulting data is in a Python dictionary object. Samples can be seen in the examples and testdata folders. JSON is the simplest way to externalize this data for further processing.

I hope this is neither too technical or too simplistic. Ask questions.

Malkierian, 2020-03-30:
LOL, too complicated is something that will probably be quite difficult to achieve. I plan on creating my own custom website for all of this. I have already created half a dozen admin-type websites in the past, so coding in PHP/HTML, etc, is no big deal. But that probably means I'll go with the REST implementation, so I can just handle everything locally myself. That InfluxDB does look like something that will be useful for this, as well, considering the overhead MySQL or PostgreSQL would add to the setup. And I already knew I'd be getting a Pi for this if the devices worked out. If I have any more specific questions about the REST implementation, I'll let you know.

Malkierian, 2020-04-04:
OK, so I finally got all the hardware to setup my permanent installation, and I've got magtest working perfectly for both RS485s. However, I can't figure out how to run the magserver example. I've used `python3 magserver.py` and using `exec` from within the python3 console. With or without help, nothing shows up on any console/terminal.

CharlesGodwin, 2020-04-04:
I'm not sure of your installation but I discovered in windows if you execute `python3` and any invalid file name, it just returns. on other platforms it gives an error. In Windows using `python` and any invalid file name gives an error.

The examples directory is not installed when you do a pip install. As they are only examples, the programs must be explicitly downloaded. Then run `python3 magserver.py` in the same directory as the downloaded file.

Malkierian, 2020-04-04:
I'm running on Raspbian now, and I already downloaded the magserver file. My terminal is in the folder I saved magserver.py to, and nothing returns, no errors.

CharlesGodwin, 2020-04-04:
i'm not sure what's wrong. To test I did this:
- I created a test folder `mkdir testmag`, then `cd testmag`, then a `git clone https://github.com/CharlesGodwin/pymagnum.git`, then `cd pymagnum/examples`
- then tested see attached log
[pilog.txt](https://github.com/CharlesGodwin/pymagnum/files/4432551/pilog.txt)

and I got a result

Malkierian, 2020-04-04:
So you need the entire repo to run magserver?

CharlesGodwin, 2020-04-04:
No, just the program. I used git as it was easier. Do you do need to install the package but since magtest worked, I assume that is installed.
Have you looked at the contents of your downloaded version of magserver.py

Charles Godwin

Malkierian, 2020-04-04:
I feel stupid now. My SFTP failed and didn't upload the changes (I copy/pasted from Github's raw display of the file).

Malkierian, 2020-04-04:
Looks like that'll be nice and easy, just gotta try to load the port as a web page. And, I do have to manually run two separate instances of it to allow for both RS485s, but that's doable.

CharlesGodwin, 2020-04-04:
You could customize magserver to read both

Charles Godwin

CharlesGodwin, 2020-04-06:
I also just realized the quick solution. Start two instances of magserver, each with a different device and different listening port. They will run concurrently as long as there is no port or device conflict.

Malkierian, 2020-04-06:
LOL, that is what I said XD

CharlesGodwin, 2020-04-06:
You're right. My bad. I've been distracted lately.  That's a reason, not an excuse

Malkierian, 2020-04-06:
No worries.

Malkierian, 2020-04-15:
Alright, finally getting to my implementation. I modified magserver.py to poll and output data from both RS485s at the same time, and I was wondering. What's the point to having the number of packets be so high? I realize just 1 or 2 packets might not get all the devices, but for my purposes, where the two inverters and the remote are the only devices on the network, would I need more than, say, 10 packets per poll?

CharlesGodwin, 2020-04-15:
No reason for high packet count. It is a variable. Make the count as small as it can be and reliably get data. If all you have are inverters that can be a very small number of packets. 50 was picked as a worse case scenario for a full blown system.

I’m glad it’s working for you.

hank, 2022-01-06:
I am going to be working with a couple MS4448 units soon in this configuration:

https://www.midnitesolar.com/productPhoto.php?product_ID=473&%20Pre%20Wired%20-%20Off%20Grid=&productCat_ID=20&act=p

Is the code currently working well for configurations like this? Sounds like I need to tap the 485 lines from somewhere other than the ARTR to prevent comma issues. Thanks so much for the above InfluxDB/Grafana details as that is what I would like to do for my in home dashboard. Do the inverters forward charge controller details like SoC, volts, load and charge rate, etc? Thanks so much for your work as it will allow me to skip the protocol decoding part I was concerned about, even though I have experience with that. Is there a list of current shortcomings?

Malkierian, 2022-01-06:
> Is the code currently working well for configurations like this?
Yes, the code is enough. I had to edit it so that it would do more than one inverter in a single application, but it is possible.
> Sounds like I need to tap the 485 lines from somewhere other than the ARTR to prevent comms issues.
If you put a splitter into the ARTR ports, and plug the inverters into one side of the splitter and the 485 leads into the other, you won't have any comms issues.
> Do the inverters forward charge controller details like SoC, volts, load and charge rate, etc?
No, you'll have to use a different utility for getting information from the Classics, if you're using Classics. If you're using others, I don't know how feasible remote routing is.
> Thanks so much for your work as it will allow me to skip the protocol decoding part I was concerned about, even though I have experience with that. Is there a list of current shortcomings?
I don't know that there are any shortcomings at the moment, besides the codebase's default inability to do multiple inverters, and lack of easy configurability. I may look into a pull request to add multiples and a config file in the near future.

CharlesGodwin, 2022-01-07:
@Malkierian Thanks for answering this. You have more experience than I with multiple inverters.

_Do the inverters forward charge controller details like SoC, volts, load and charge rate, etc?_ 

The inverter interface provides many values relative to inverter data such as AC volts and amps, DC volts and amps from inverter's perspective as well as status values for inverter on, charging, etc. In the system you referenced there is also a Magnum BMK (Battery Monitor Kit) which my code also supports. It provides values such as SOC, battery voltage,
and DC charge/discharge rate. As @Malkierian  said, the details of the Classic charge controller are a separate utility and I cannot help you.

@Malkierian  Concerning your comments about the codebase. Can you open a discussion on this [here ](https://github.com/CharlesGodwin/pymagnum/discussions) and I'll add my thoughts.

CharlesGodwin, 2022-01-23:
@Malkierian 
I have written a proposed solution for multiple devices as a [discussion ](https://github.com/CharlesGodwin/pymagnum/discussions)
I welcome your input or an alternative solution.

I plan another discussion about configuration file(s). This is more complicated to implement. Stay tuned.

Thanks