CharlesGodwin/pymagnum #10: OSError: [Errno 25] Inappropriate ioctl for device

vance4c, 2020-05-21:
I have a fresh install of raspbian on a RPi W. Installed from pip3 and the run of magtest I get this output. Am I missing a package? Do you have a suggestion to fix this issue? Thank you for making this package I am really looking forward to using it.

root@raspberrypi:/home/pi# magtest -d /dev/ttyAMA0 --timeout 0.005
Version:1.1.0
Options:cleanpackets=True, device='/dev/ttyAMA0', packets=50, timeout=0.005
Error detected attempting to read network data - test failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/magnum/tools/magtest.py", line 47, in main
    packets = reader.getPackets()
  File "/usr/local/lib/python3.7/dist-packages/magnum/magnum.py", line 124, in getPackets
    packets = self.readPackets()
  File "/usr/local/lib/python3.7/dist-packages/magnum/magnum.py", line 164, in readPackets
    if self.reader.inWaiting() == 0:
  File "/usr/lib/python3/dist-packages/serial/serialutil.py", line 590, in inWaiting
    return self.in_waiting
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 467, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
OSError: [Errno 25] Inappropriate ioctl for device

CharlesGodwin, 2020-05-21:
Thank you for your interest in my software. I hope we can resolve this problem.

What RS485 Device did you install?
If it is a HAT type, did you install it according to the manufacturer's instructions including any required changes / additions to system software?

I just  tested on a new install with no adjustments for a HAT and got the same result as you have reported.

If I knew which hat you installed may be able to recommend changes. A RS485 USB dongle doesn't have this problem.

CharlesGodwin, 2020-05-23:
@vance4c Have you had any luck finding the solution? can I help?

vance4c, 2020-05-26:
Thank you for your response. Sorry for taking a while the email
notification was routed weird.

I am using a hat. This is the one I bought:
https://www.waveshare.com/rs485-can-hat.htm

I found on their site the modules to add to the /boot/config.txt file.

CharlesGodwin, 2020-05-26:
I have the same device, installed it last year and have no problems with it.  I recommend you double check your setup.

I just tested with it. I'm not connected to my magnum network but I got this result.

`magtest --device /dev/ttyAMA0
Version:1.1.0
Options:cleanpackets=True, device='/dev/ttyAMA0', packets=50, timeout=0.001
Error detected attempting to read network data - test failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/magnum/tools/magtest.py", line 47, in main
    packets = reader.getPackets()
  File "/usr/local/lib/python3.7/dist-packages/magnum/magnum.py", line 124, in getPackets
    packets = self.readPackets()
  File "/usr/local/lib/python3.7/dist-packages/magnum/magnum.py", line 165, in readPackets
    raise ConnectionError("There doesn't seem to be a network")
ConnectionError: There doesn't seem to be a network`

But at least it tried.

vance4c, 2020-05-26:
I went back and double checked the setup here:https://www.waveshare.com/w/upload/2/29/RS485-CAN-HAT-user-manuakl-en.pdf

I was missing the python-can library for the example code. 
Then I went back to your setup page and and tried /dev/ttyS0 and did not get the error this time.
but I do with /dev/ttyAMA0.
I do not get packet get from the magtest. I am starting the troubleshooting soon.

CharlesGodwin, 2020-05-26:
You can list all possible serial ports with this command
` python3 -m serial.tools.list_ports`

some people get different results. Test each in turn. Check your magnum wiring. The first suspect is the rj11 phone cable, does it have 4 connectors. Magnum uses the two outside wires and normal phones use the inside two.

vance4c, 2020-05-26:
This is what I get with the command you sent. I still get the error message
when I use this serial port. I will keep checking my setup. I really
appreciate your help. I am a heavy Linux user but this is my first hardware
hat so I don't put it past me to have missed something,

root@raspberrypi:/home/pi/Scripts/RS485_CAN_HAT_Code/485/python# python3 -m
serial.tools.list_ports
/dev/ttyAMA0
1 ports found

vance4c, 2020-05-26:
I made my own cables. They have 6 connectors and in a reverse configuration. 1->8, 2->7, 3->6 ... I found this mapping in a Magnum manual. However like you posted before it should try and to connect without a network attached.

CharlesGodwin, 2020-05-26:
OK. I need to do some memory work on this. It may be a while.

I know it says only the one. But what if you try /dev/ttyS0?  

Others have had problems but its usually a wiring problem not software. But eho know, it could be anything.

CharlesGodwin, 2020-05-26:
Even with nothing connected you should at least get to `ConnectionError: There doesn't seem to be a network` Then play with the wiring. I've never seen a Magnum wiring diagram with 6 wires, usually just 4. I do know on a 4 wire they use the 1 and 4. Sometimes their cables crossover, sometimes they don't.

vance4c, 2020-05-26:
here is the screenshot from the manual I found. I made both a straight and a reverse cable and plugged them into my remote and the inverter ant the reverse worked.

[image]

CharlesGodwin, 2020-05-26:
There are two problems.

1. Are you still getting `OSError: [Errno 25] Inappropriate ioctl for device` problems? That's software, not a cable. I have asked some one with experience for their input.
2. Are you getting 'ConnectionError: There doesn't seem to be a network`? That's a wiring issue.

If it's a wiring issue and you are using 6 conductor wiring then be sure you are working with pin 2 and pin 5 for the HAT connection.

vance4c, 2020-05-26:
The first option [Errno 25]
[image]

CharlesGodwin, 2020-05-26:
Which Pi? 3 or 4

Which version of OS?

I'm running out of ideas.

Charles Godwin

[Annotation 2020-05-26 145630]<https://user-images.githubusercontent.com/65700471/82954260-3353de80-9f61-11ea-87e1-b60729011629.png>

vance4c, 2020-05-26:
Which Pi? Raspberry Pi Zero W  ( I have a RPi 1 and a RPi 3 as well I could test with)

Which version of OS?
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian

CharlesGodwin, 2020-05-26:
Thanks I'll keep thinking. I may build one to test

Charles Godwin

vance4c, 2020-05-27:
Just to be sure I attached it to my RPi 3 and still have the same error. I might reload from scratch and try again.

CharlesGodwin, 2020-05-27:
Tomii an to build a pi zero w and test

Charles Godwin

vance4c, 2020-05-27:
I have reinstalled Raspbian buster and Raspbian stretch on my RPi 3 with the same error both times.

CharlesGodwin, 2020-05-27:
What process did you use use to install WiringPi? I find confusing instructions in the WaveShare instructions. Initial attempts to build have failed.

CharlesGodwin, 2020-05-27:
I got good results. I followed the instructions very carefully. Here's what I did.

- Used new Buster Lite image on Pi 3 B+
- Initial setup - raspi-config
	- changed host name
	- changed locale
	- changed timezone
	- changed keyboard
	- changed wifi country
- full upgrade
- took snapshot of /boot/config.txt
- Installed wiringpi `sudo apt-get install wiringpi`
- tested install
<pre>
gpio -v
gpio version: 2.50
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 3B+, Revision: 03, Memory: 1024MB, Maker: Sony
  * Device tree is enabled.
  *--> Raspberry Pi 3 Model B Plus Rev 1.3
  * This Raspberry Pi supports user-level GPIO access.
</pre>
- Used install manual RS485-CAN-HAT-user-manuakl-en.pdf
- modified /boot/config.txt  added 
<pre>dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=8000000,interrupt=25,spimaxfrequency=1000000</pre>
- raspi-config - again 
  - disable select shell over serial port
   - enable serial port
- reboot
- test per install manual
<pre>dmesg | grep -i '\(can\|spi\)'
[    6.009260] CAN device driver interface
[    6.150234] mcp251x spi0.0 can0: MCP2515 successfully initialized</pre>

- installed pip3 `sudo apt-get install python3-pip`
- Installed Magnum software `sudo pip3 install pymagum`
- tested `magtest --device /dev/ttyS0` Succesful
- Why `/dev/ttyS0`? No idea, but it is the port used in the manual for testing and it works. `/dev/ttyAMA0` does not.

vance4c, 2020-05-27:
I do not get an error with /dev/ttyS0 I'll hook it back up and see it the network is working.

CharlesGodwin, 2020-05-27:
Good luck. I wish you well

Charles Godwin

vance4c, 2020-05-30:
I have good news and bad new. I connected it back up and I do not get errors! The magtest works however I get 25 out of 50 unknown packets. I switched the wires and then I get nothing. Next I will switch it back and change the timing. Thank you for you'r incredible help with this!

CharlesGodwin, 2020-05-30:
Glad I've helped but I can't wait for success. Try a smaller timeout. It may help

vance4c, 2020-06-01:
Here is the output with 0.005 timeout. Do you have other suggestions what I am missing? 

root@PowerPi:/home/pi# magtest -d /dev/ttyS0 -t 0.005
Version:1.1.0
Options:cleanpackets=True, device='/dev/ttyS0', packets=50, timeout=0.005
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000066B3008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B2008022232E292323FE
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000068B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000068B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000066B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000066B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B3008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B3008022232E292323FE
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B3008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B3008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000067B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000066B3008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000066B3008022232E292323FE
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000083000067B3008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000065B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000065B2008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000068B3008022232E292323FF
Length:33 UNKNOWN   =>000A1404641E0C00006E6E000000112302000082000068B3008022232E292323FF
Packets:25 of 50 with 25 UNKNOWN, in 2.99 seconds

CharlesGodwin, 2020-06-01:
Try 
`magtest --device /dev/ttyS0  --nocleanup --timeout 0.001`

then, if it still looks bad slowly step the time out up by 0.001 increments

`magtest --device /dev/ttyS0  --nocleanup --timeout 0.002`

but stop at

`magtest --device /dev/ttyS0  --nocleanup --timeout 0.010`

If you're still having problems, and if it's not too difficult, please send me the outputs.

Also please tell me what devices you have connected in the network, especially the Inverter's exact model number.

Thanks for your patience

vance4c, 2020-06-01:
Here are the logs.
[magtest.log](https://github.com/CharlesGodwin/pymagnum/files/4713143/magtest.log)

CharlesGodwin, 2020-06-01:
Thanks

Can you rerun  the 0.005 and 0.010 and include the `--nocleanup` option.

`magtest --device /dev/ttyS0 --nocleanup --timeout 0.010`

Also what equipment do you have.

Thanks

CharlesGodwin, 2020-06-01:
@vance4c  I also notice in the first log in the file is a 0.002 timeout. Do you have the 0.001 version?

Here are the first 4 lines in the file I got. Line 3 is a display, from the program, of the options selected.

<pre>
--nocleanup --timeout 0.001
Version:1.1.0
Options:cleanpackets=False, device='/dev/ttyS0', packets=50, timeout=0.002
Length:17 UNKNOWN   =>02000082000067B3008022232E292323FF
</pre>

Thanks

CharlesGodwin, 2020-06-01:
Sorry. I missed the info about the equipment in the earlier message. I see you just added it. Just as I saw it online I had guessed your setup from the --nocleanup log.

I know the problem. i never wrote code for these older versions of inverter and remote. Many years ago, Magnum revised their packet definitions and made them longer. I never considered the older units active. I have the specifications for these components but I can't give you an estimate when I'll have changes ready for testing. It may be a week or two.

Can I still have a 0.001 timeout log with --nocleanup set? It will help me test my new code.

CharlesGodwin, 2020-06-01:
@vance4c 
Can I also have the model and revision of the Remote? The revision may be available from the LCD display. It is on newer models.

vance4c, 2020-06-01:
root@PowerPi:/home/pi# magtest -d /dev/ttyS0 --nocleanup --timeout 0.001
Version:1.1.0
Options:cleanpackets=False, device='/dev/ttyS0', packets=50, timeout=0.001
Length:17 UNKNOWN   =>02000081000067B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000067B300802227322F2323FE
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000082000066B200802227322F2323FE
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000082000066B200802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B200802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B200802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B200802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B200802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000065B300802227322F2323FE
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000065B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000065B200802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000082000066B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000082000066B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B200802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B200802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B200802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000066B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000064B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000081000064B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Length:17 UNKNOWN   =>02000082000065B300802227322F2323FF
Length:16 UNKNOWN   =>000A1404641E0C00006E6E000000171F
Packets:50 of 50 with 50 UNKNOWN, in 2.99 seconds

CharlesGodwin, 2020-06-01:
Thanks. Now to work

Charles Godwin

vance4c, 2020-06-01:
The remote is 1.2
Inverter very 3.4

CharlesGodwin, 2020-06-02:
Thanks

Charles Godwin

CharlesGodwin, 2020-06-06:
@vance4c  Just an update. I plan to start work on this bug June 7 or June 8. I don't think it will take too much time.

vance4c, 2020-06-06:
Thank you. I am impressed with your dedication and support of this project.

CharlesGodwin, 2020-06-07:
@vance4c 

A question. When you ran the tests for the log, was the inverter connected to external AC so it would be in charge mode? If yes, could you run another set without the AC-in so it is using the inverter?
`magtest --device /dev/ttyS0 --nocleanup --timeout 0.001`

thanks

vance4c, 2020-06-07:
Yes AC was on. I'll run the test when I can soon.

CharlesGodwin, 2020-06-07:
Great to hear. That's what the data suggested. Using your previous data and my test code I get this. does it look reasonable?
[issue_10.txt](https://github.com/CharlesGodwin/pymagnum/files/4742612/issue_10.txt)

vance4c, 2020-06-08:
Here is the logs. I did a combination of charging and inverting with load
and without.

CharlesGodwin, 2020-06-08:
@vance4c 
I don't see the logs.

vance4c, 2020-06-08:
Sorry I tried to attach it to the email.

[test2.log](https://github.com/CharlesGodwin/pymagnum/files/4746927/test2.log)

CharlesGodwin, 2020-06-08:
Thanks. Systems don't always do what we think they should.

I separated each run into a separate file for testing. the json file is the output. it all makes sense but the Ac-in and AC-out numbers seem odd. What do you  think?  Am I close? done? Want code to test on your site?

[samples.zip](https://github.com/CharlesGodwin/pymagnum/files/4747178/samples.zip)

vance4c, 2020-06-08:
This is amazing. Thank you for you hard work. I am looking at sample 5 and the adc is exactly what the remote was indicating. So that looks great. I also thought AC-in / AC-out looked weird too. I assume these are the voltage ac in and out? it is connected to AC 220 both L1 and L2 also 220 out both L1 and L2.

CharlesGodwin, 2020-06-08:
That's encouraging. Want a version to install?

vance4c, 2020-06-08:
Are your changes posted to Github in a different branch? Or is this a completly separate project?

CharlesGodwin, 2020-06-08:
Yes they are, in improve-testing-tools (I'm not that good on multiple branches) but you'd have to build and install it. If you want a pip compatible install, let me know and in 2 minutes it will be available on the web from a test site. A trivial exercise for me.

vance4c, 2020-06-08:
I measured and the AC-in is 245 V ish L1 and L2 around 121 V. with and AC out the same. With ac-in off the inverting ACV out is 121 on both L1 and L2. However the I get no voltage difference between L1 and L2 on whe inverting. Is that normal?

CharlesGodwin, 2020-06-08:
I'm not an expert so all of this is a layman's opinion. I believe it all depends on how the inverter is wired. I know with mine I could chose between high current all 120VAC or the conventional 120/240 VAC setup. For you device refer to the owners manual [here](https://www.magnum-dimensions.com/sites/default/files/product/manual/64-0007%20Rev%20F%20%28MS%20Series%29_Web_1.pdf) and look at Table 2-3, AC Input/Output Wiring Configurations.

The short answer: Probably normal.

vance4c, 2020-06-08:
I have not made a package before. I am willing to learn or a built package would be nice.

vance4c, 2020-06-08:
Ok yes that lead me to the answer it is 240VAC in standby only. It will not invert 240 but it will pass it through from the shore power. Thank you again. I have read that manual s many times and missed it.

CharlesGodwin, 2020-06-08:
That's the thing about manuals. Until you have a question they don't provide an answer. ;)

Re the build
you can install from my test site with
`@echo sudo pip3 install --upgrade -i https://test.pypi.org/simple/ pymagnum=1.1.1b4

vance4c, 2020-06-08:
Here is the first magtest

Version:1.1.1b4
Options:cleanpackets=True, device='/dev/ttyS0', packets=50, timeout=0.001, trace=False
Length: 1 UNKNOWN   =>FF
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000067B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000084000068B2008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000084000068B2008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000084000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000084000068B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000084000068B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>0200008400006AB3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>0200008400006AB3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000068B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000068B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000067B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000067B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000067B2008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000067B2008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000084000068B2008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Length:16 INVERTER_C=>02000083000069B3008022212C282323
Length:16 REMOTE_C  =>000A1404641E0C00006E6E0000001306
Packets:50 of 50 with 1 UNKNOWN, in 2.90 seconds

CharlesGodwin, 2020-06-08:
That's the thing about manuals. Until you have a question they don't provide an answer. ;)

Re the build
you can install from my test site with
`sudo pip3 install --upgrade -i https://test.pypi.org/simple/ pymagnum=1.1.1b4`

or, the slow way,  
- download the whole git repository
- run this once
`sudo pip3 install -r requirements.txt`
- run the windows batch file included `build_draft.cmd`, or if on a Pi, run lines 3 and 4
- run `sudo pip3 install dist/pymagnum-1.1.1b4.zip`

That still won't tell you how to define a package, just how build this one.

CharlesGodwin, 2020-06-08:
try magdump now. It will spit out a json record. Use same parameters

vance4c, 2020-06-08:
{"datetime":"2020-06-08T12:09:05-07:00","device":"MAGNUM","data":[{"device":"INVERTER","data":{"revisio             n":"3.4","mode":2,"mode_text":"FLOAT","fault":0,"fault_text":"None","vdc":13.1,"adc":0,"VACout":105,"VA             Cin":178,"invled":0,"invled_text":"Off","chgled":1,"chgled_text":"On","bat":33,"tfmr":44,"fet":40,"mode             l":35,"model_text":"MS2012","stackmode":0,"stackmode_text":"Stand Alone"}},{"device":"REMOTE","data":{"             revision":1.2,"action":0,"searchwatts":10,"batterysize":240,"battype":4,"absorb":0,"chargeramps":100,"a             input":30,"parallel":0,"force_charge":0,"lbco":11.0,"vaccutout":110}}]}

vance4c, 2020-06-08:
I don't know what all the entries are but looking very good. VACout and VACin still are funky.

CharlesGodwin, 2020-06-08:
if you want it pretty
`magtest ......... |python3 -m json.tool`

The ... are the parameters

I'm not sure what to say about AC in and out. It's what's in the data.

I just reviewed their documentation and I quote 
"The data is not very accurate
0-150 Volts ac RMS Meter is better with newer Software but stil not great"

[4 more replies]

closed 2020-06-08 by CharlesGodwin