CharlesGodwin/pymagnum #16: Erratic behaviour at boot time

CharlesGodwin, 2020-08-09:
sometimes, If the magnum reader is invoked shortly  after startup, random voltage is applied to the RS-485 device. This can cause unexpected behaviour of devices on the Magnum network, including the inverter.

This will be fixed in release 1.2 by enforcing a 30 second delay from boot time to ensure a stable system.

ve2mxo, 2021-05-05:
The application put the com ports in error state, no clue how to fix it.
To be able to get data, i reboot the RPI every 15 minutes an start magdump and magtest in a batch at startup.
The sequence seems to send a toggle to switch the inverter on/off.
The systmem id 400km away the last message show :
Master : invert or OFF with an error code of 127 (no reference about it.
Slave : Standby or off no error message
[magdump0.log](https://github.com/CharlesGodwin/pymagnum/files/6424645/magdump0.log)
[magdump1.log](https://github.com/CharlesGodwin/pymagnum/files/6424646/magdump1.log)
[magdump2.log](https://github.com/CharlesGodwin/pymagnum/files/6424636/magdump2.log)

CharlesGodwin, 2021-05-05:
I seem to have let this slip. I apologize. I'll get to work on a built in delay.

I don't know the specifics of how you are running but could you try adding `sleep 30s` to your batch before the magnum program to let stuff settle down. I have had success with that technique. But once, a few years ago, I had a defective RS485/USB dongle. I replaced it and problems went away.

CharlesGodwin, 2021-05-05:
I do have the code for the next release. I'll get busy publishing a beta copy for you to test.

ve2mxo, 2021-05-05:
There is two problems:
1=>systematic lost of communication on the ports, not able to reconnect
before a reboot.
2=>Erratic toggling of the inverter, I already wait some time before
starting the mag* utilities.
I use the official version of magnumpy.

CharlesGodwin, 2021-05-05:
I'm at  loss for a solution. you could try my next release that is still being tested.
run this
`sudo pip3 install pymagnum==2.0rc1`

if that doesn't help or you want to remove it do these two steps.
`sudo pip3 uninstall pymagnum`
and
`sudo pip3 install pymagum`

What type of RS485 device are you using? Can you test with a different one?

CharlesGodwin, 2023-02-19:
I added a start-up delay to version 2.0 and no one has reported more problems.

closed 2023-02-19 by CharlesGodwin