CharlesGodwin/pymagnum #20: Discussions is now open.

CharlesGodwin, 2021-03-10:
I've opened the [discussion ](https://github.com/CharlesGodwin/pymagnum/discussions)feature so Q&A etc can go there and this can remain only for issues and bugs.

rwwende, 2021-03-24:
I'm curious how MQTT gets the feed from your program.  Sorry very new at this.  thanks

CharlesGodwin, 2021-03-24:
In the examples folder in this repository is an example program named mqttlogger.py. It runs a loop that reads the RS485 data and send JSON payloads to a MQTT broker (server). Thi s can be modified / shaped to suit your MQTT needs.

rwwende, 2021-03-24:
> CAUTION:
>  This email originated outside of Lumen Technologies. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Thank you!

Ross Wendell

This communication is the property of Lumen Technologies and may contain confidential or privileged information. Unauthorized use of this communication is strictly prohibited and may be unlawful. If you have received this communication in error, please immediately notify the sender by reply e-mail and destroy all copies of the communication and any attachments.