CharlesGodwin/pymagnum #21: Magnum Inverter weird VAC in reading

tomtdl, 2021-07-27:
Hi Charles,
                   I appreciate all you have done for the community and we users of Magnum products.

When I run magserver or magsample with my single MS-4448PAE and ME-RTR, All readings are in agreement with the RTR display and believable.

The  VAC IN  though , at 171 or 172VAC does not reflect the actual measured input of 108 to 115VAC per leg, (or 216 - 230VAC total).  It is somewhere in between.  

1)  Has anyone else reported similar findings?  When I read the logs others have posted, the VAC IN is usually 0 (must be in Invert mode - no AC IN connected?)  Any idea why this one field is wrong?

2)  On a different tangent,  I am working on an Arduino web interface to this data thru ethernet shield.  Will happily share any of my efforts in this area.
Also doing an ESP8266 Wifi Web page or UDP packet transfer.

3)  Does anyone know if there is an easy way to toggle between INVERT and PASS-THROUGH, say without changing trip point settings in the RTR?  My experience with the Outback inverters suggests this may available without interrupting the AC IN or changing the trip points, as they have a USE or DROP on their remote, and on their web interface.

Again, thank you for all your help to us.

Tom Leitschuh

CharlesGodwin, 2021-07-27:
Thanks for the feedback.

The AC IN field, and AC Input Amps field are indeed for AC Connected, either through main power or a running generator. In off-grid situations these are at 0 except when generator is running. 

Concerning the voltage value. Here is a quote directly from the Magnum documentation I used to set up my software describing the field "0-255 Volts ac peak Meter is very bad".  The first part of that is important. Alternating Current power is a sine-wave and produces a continuously changing voltage. Most electrical meters use a technique called RMS to measure the average voltage, which is 120VAC. However the peak voltage is higher. This peak value is what the inverter is reporting. Alas, I can't fix that as I am only reporting data generated the Magnum equipment. As noted in the documentation, "is very bad".

I'm sure others would be interested in your Arduino implementation. I suggest you create  a GitHub repository for your work. I will add a reference to your repository here when it is ready.

concerning part 3. I have no idea, maybe others can help.

tomtdl, 2021-07-27:
Charles,
              I believe you are right about the peak voltage.  When I multiply it by 0.707, it matches the true RMS meter reading.  They must be using a different method for the AC OUT voltage, as that matches my meter without correction, even as it will vary under load.  
Thank you for pointing that out.  My load power factor on both legs is 0.95 to 1.00.  I was hoping someone who is on grid or generator could report their AC IN readings.  Again, I appreciate your work on this project.

Tom

CharlesGodwin, 2021-07-28:
I'm not at my system before tomorrow. I'll check my logs then about generator voltage

closed 2023-02-22 by CharlesGodwin