CharlesGodwin/pymagnum #33: Inappropriate ioctl for device

rjdwadev, 2022-05-29:
Hi - when running magtest I am receiving the following error; any ideas?

`Error detected attempting to read network data - test failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/magnum/tools/magtest.py", line 49, in main
    packets = reader.getPackets()
  File "/usr/local/lib/python3.9/dist-packages/magnum/magnum.py", line 175, in getPackets
    packets = self.readPackets()
  File "/usr/local/lib/python3.9/dist-packages/magnum/magnum.py", line 215, in readPackets
    if self.reader.inWaiting() == 0:
  File "/usr/lib/python3/dist-packages/serial/serialutil.py", line 594, in inWaiting
    return self.in_waiting
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 531, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
OSError: [Errno 25] Inappropriate ioctl for device
`

CharlesGodwin, 2022-05-29:
It could be several things. Let's start with the simple ones first. Please do the following: 

- Tell me what machine, OS and version are you using?
- run a test program to determine where your RS485 device may be. run this `python3 -m serial.tools.list_ports`
- what command did you run for magtest?
- Are you sing a RS485 HAT or a RS$*%/USB dongle?

rjdwadev, 2022-05-30:
Hi Charles --  thank you for the reply, here's the info:

* Tell me what machine, OS and version are you using? --> I'm using a Canakit Pi 4/ 4gb ram -- I'm running the bullseye linux 11 OS release
* run a test program to determine where your RS485 device may be. run this python3 -m serial.tools.list_ports --> I'm using a RS485 HAT, the port that is returned is ttyAMA0
* what command did you run for magtest? --> I ran magtest -d /dev/ttyAMA0
* Are you sing a RS485 HAT or a RS$*%/USB dongle?--> RS485 HAT

CharlesGodwin, 2022-05-30:
I’m researching, but while you wait try these: (on a hunch only)

- sudo magtest -d /dev/ttyAMA0
- magtest -d /dev/ttyS0
- sudo magtest -d /dev/ttyS0

CharlesGodwin, 2022-05-30:
Try this, 

- sudo raspi-config
- select option 3 - Interface Options
- select option I6 - Serial Port
- select No to  Would you like a login shell to be accessible over serial?
- select yes to │ Would you like the serial port hardware to be enabled?

This will require a reboot

Then repeat your tests.

rjdwadev, 2022-05-30:
Thanks Charles - I completed the steps above, and ran the tests you listed above - running sudo magtest -d /dev/ttyAMA0 results in the same error. However, running magtest -d /dev/ttyS0, or sudo magtest -d /dev/ttyS0 results in the below - which I think indicates that I have a cross-wiring problem?

`Version:1.1.3
Options:device='/dev/ttyS0', packets=50, timeout=0.001, cleanpackets=True, trace=False
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF01FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFBF00FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE7FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF03
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE7FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF00FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF01FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF07
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF01FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF01FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF03
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF01FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFBF00FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE9FFFF0DFFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE9FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE9FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF00FDFDE9FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF01FDFDE9FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF03
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF01FDFDE9FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF01FDFDE7FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9FFFFAFFFFF01FDFDE7FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFBF00FDFDE9FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE9FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF01
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE9FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF07
Length:41 UNKNOWN   =>F5AFEF5FC3C7F3FF6F32CF5ED5A9D7FF23FFFF01FDFDE9FFFF11FFFDFF91D1CBD52DFFFFFFFB4FFF01
Packets:25 of 50 with 25 UNKNOWN, in 3.17 seconds
`

rjdwadev, 2022-05-30:
Yes! success!! Thank you Charles - I'm now getting: 

`
Version:1.1.3
Options:device='/dev/ttyS0', packets=50, timeout=0.001, cleanpackets=True, trace=False
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_80 =>00052808501E1C0600C89B840C0F15360000280080
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F15360000280000
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F15360000280000
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F15360000280000
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F15360000280000
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_A0 =>00052808501E1C0600C89B840C0F153614006E00A0
Length:21 INVERTER  =>4000010C00007700010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_80 =>00052808501E1C0600C89B840C0F15360000280080
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F15360000280000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F15360000280000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F15360000280000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F15360000280000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_A0 =>00052808501E1C0600C89B840C0F153614006E00A0
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_00 =>00052808501E1C0600C89B840C0F153614006E0000
Length:21 INVERTER  =>4000010C00007800010037171A1569000000025800
Length:21 REMOTE_80 =>00052808501E1C0600C89B840C0F15360000280080
Packets:50 in 3.12 seconds
`

rjdwadev, 2022-05-30:
By the way, here is the HAT I am using: https://www.amazon.com/RS485-CAN-HAT-Long-Distance-Communication/dp/B07VMB1ZKH/ref=pe_386300_442618370_qpr_2/130-9926171-3959568/ref=qpr?_encoding=UTF8&pd_rd_i=B07VMB1ZKH&psc=1&pd_rd_w=qPksl&pf_rd_p=2411a357-51fe-4cde-ba0e-3cb2d30291e8&pf_rd_r=2ZZE05TQ0C0JH8R6D7TP&pd_rd_r=701859e6-9301-4569-8289-df76b4e5ed38&pd_rd_wg=rclKx

CharlesGodwin, 2022-05-31:
Great. you can also try magdump and you'll get a JSON result. I see you have only an Inverter and a Remote. Is that correct?

CharlesGodwin, 2022-06-02:
Can I close this?

CharlesGodwin, 2022-06-03:
Closing. Assumed fixed.

closed 2022-06-03 by CharlesGodwin

rjdwadev, 2022-06-04:
Apologies Charles -- yes, this is good to close. Thank you for your help!!

CharlesGodwin, 2022-06-04:
I'm just glad I can help and it's useful.

rjdwadev, 2022-10-11:
Apologies Charles - yes, I think I am all set for right now. Thank you!!